jcmvbkbc/crosstool-NG #11: Compiling Error: Build failed in step 'Installing final compiler'

s137, 2015-05-03:
I tried building the esp-open-sdk from pfalcon, which includes your crosstool-NG, in order to compile nodemcu for my ESP.

I got the following error on my Ubuntu 10.04.4 32bit System, during the building process:

```bash
[INFO ]  Installing final compiler
[ERROR]    make[4]: *** [cc1] Error 1
[ERROR]    make[3]: *** [all-gcc] Error 2
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing final compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_cc_core_backend[scripts/build/cc/gcc.sh@483]
[ERROR]  >>        called from: do_cc_for_host[scripts/build/cc/gcc.sh@595]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@638]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 53:36.00)
[53:37] / make[2]: *** [build] Fehler 2
make[2]: Verlasse Verzeichnis '/home/simon/esp-open-sdk/crosstool-NG'
make[1]: *** [_toolchain] Fehler 2
make[1]: Verlasse Verzeichnis '/home/simon/esp-open-sdk/crosstool-NG'
make: *** [/home/simon/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Fehler 2
```

I hope you can help me with this.
Thanks in Advance.
s137

jcmvbkbc, 2015-05-03:
Please share the file /home/simon/esp-open-sdk/crosstool-NG/build.log

s137, 2015-05-03:
Uhm... the file you are requesting contains 136981 lines of text... and my firefox crashes if i try to put this in here, so i give you only the last piece of output:

```
[ALL  ]    No forbidden identifiers defined by this target
[ALL  ]    Finding directories and links to directories
[ALL  ]     Searching /home/simon/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/.
[ALL  ]    Making symbolic directory links
[ALL  ]    Fixing directory /home/simon/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include into /home/simon/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/gcc/include-fixed
[ALL  ]    Cleaning up unneeded directories:
[ALL  ]    fixincludes is done
[ALL  ]    echo timestamp > stmp-fixinc
[ALL  ]    if [ -d include ] ; then true; else mkdir include; chmod a+rx include; fi
[ALL  ]    if [ -d include-fixed ] ; then true; else mkdir include-fixed; chmod a+rx include-fixed; fi
[ALL  ]    for file in .. /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/ginclude/float.h /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/ginclude/iso646.h /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/ginclude/stdarg.h /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/ginclude/stdbool.h /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/ginclude/stddef.h /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/ginclude/varargs.h /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/ginclude/stdfix.h /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/ginclude/stdnoreturn.h /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/ginclude/stdalign.h /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/ginclude/tgmath.h; do 	  if [ X$file != X.. ]; then 	    realfile=`echo $file | sed -e 's|.*/([^/]*)$|1|'`; 	    echo timestamp > include/$realfile; 	    rm -f include/$realfile; 	    cp $file include; 	    chmod a+r include/$realfile; 	  fi; 	done
[ALL  ]    for file in .. ; do 	  if [ X$file != X.. ]; then             mv include/$file include/x_$file;             echo "#include_next <$file>" >include/$file;             cat include/x_$file >>include/$file;             rm -f include/x_$file; 	    chmod a+r include/$file; 	  fi; 	done
[ALL  ]    for file in .. ; do 	  if [ X$file != X.. ]; then 	    echo "#include_next <$file>" >>include/$file; 	    chmod a+r include/$file; 	  fi; 	done
[ALL  ]    rm -f include/stdint.h
[ALL  ]    if [ wrap = wrap ]; then 	  rm -f include/stdint-gcc.h; 	  cp /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/ginclude/stdint-gcc.h include/stdint-gcc.h; 	  chmod a+r include/stdint-gcc.h; 	  cp /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/ginclude/stdint-wrap.h include/stdint.h; 	  chmod a+r include/stdint.h; 	elif [ wrap = provide ]; then 	  cp /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/ginclude/stdint-gcc.h include/stdint.h; 	  chmod a+r include/stdint.h; 	fi
[ALL  ]    set -e; for ml in `cat fixinc_list`; do 	  sysroot_headers_suffix=`echo ${ml} | sed -e 's/;.*$//'`; 	  multi_dir=`echo ${ml} | sed -e 's/^[^;]*;//'`; 	  fix_dir=include-fixed${multi_dir}; 	  if [ -f `echo /home/simon/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot${sysroot_headers_suffix}/usr/include | sed -e :a -e 's,[^/]*/../,,' -e ta`/limits.h ] ; then 	    cat /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/limitx.h /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/glimits.h /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/limity.h > tmp-xlimits.h; 	  else 	    cat /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/glimits.h > tmp-xlimits.h; 	  fi; 	  /bin/bash /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../mkinstalldirs ${fix_dir}; 	  chmod a+rx ${fix_dir} || true; 	  /bin/bash /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../move-if-change 	    tmp-xlimits.h  tmp-limits.h; 	  rm -f ${fix_dir}/limits.h; 	  cp -p tmp-limits.h ${fix_dir}/limits.h; 	  chmod a+r ${fix_dir}/limits.h; 	done
[ALL  ]    rm -f include-fixed/README
[ALL  ]    cp /home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../fixincludes/README-fixinc include-fixed/README
[ALL  ]    chmod a+r include-fixed/README
[ALL  ]    echo timestamp > stmp-int-hdrs
[ALL  ]    i686-build_pc-linux-gnu-ranlib  libbackend.a
[ALL  ]    build/genchecksum c/c-lang.o c-family/stub-objc.o attribs.o c/c-errors.o c/c-decl.o c/c-typeck.o c/c-convert.o c/c-aux-info.o c/c-objc-common.o c/c-parser.o c-family/c-common.o c-family/c-cppbuiltin.o c-family/c-dump.o c-family/c-format.o c-family/c-gimplify.o c-family/c-lex.o c-family/c-omp.o c-family/c-opts.o c-family/c-pch.o c-family/c-ppoutput.o c-family/c-pragma.o c-family/c-pretty-print.o c-family/c-semantics.o c-family/c-ada-spec.o tree-mudflap.o default-c.o libbackend.a main.o  libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a libcommon.a ../libcpp/libcpp.a ../libiberty/libiberty.a   ../libdecnumber/libdecnumber.a ../libbacktrace/.libs/libbacktrace.a                      checksum-options > cc1-checksum.c.tmp && 		 	/home/simon/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../move-if-change cc1-checksum.c.tmp cc1-checksum.c
[... cut ...]
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_cc_core_backend[scripts/build/cc/gcc.sh@483]
[ERROR]  >>        called from: do_cc_for_host[scripts/build/cc/gcc.sh@595]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@638]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]  
[ERROR]  (elapsed: 53:36.00)h
```

jcmvbkbc, 2015-05-03:
> ld terminated with signal 9 [Killed]

Looks like you don't have enough memory to link the compiler. Please consider using more swap space or machine with more physical memory.

s137, 2015-05-03:
Alright thanks, do you think 512MB of RAM in combination with 2GB of Swap space is enough?

jcmvbkbc, 2015-05-03:
Depends on how much is actually free. 2.5GB of free RAM+swap sounds like enough.

closed 2015-05-18 by jcmvbkbc