jcmvbkbc/crosstool-NG #14: ERROR in build -- [ALL  ]    *** Newlib does not support CPU xtensa

DrObscure, 2015-08-04:
In trying to build the tree in FreeBSD 10.1, I am getting the following errors about the Newlib lib..
The makefile is set to use the gnu tools (gmake,gsed,gawk,etc)

Here is the last few lines of build.log:
[ALL  ]    checking dependency style of xtensa-unknown-elf-cc -B/research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-unknown-elf/build/build-libc/xtensa-unknown-elf/newlib/ -isystem /research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-unknown-elf/build/build-libc/xtensa-unknown-elf/newlib/targ-include -isystem /research/Esp8266/esp-open-sdk/crosstool-NG/.build/src/newlib-1.20.0/newlib/libc/include -B/research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-unknown-elf/build/build-libc/xtensa-unknown-elf/libgloss/xtensa -L/research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-unknown-elf/build/build-libc/xtensa-unknown-elf/libgloss/libnosys -L/research/Esp8266/esp-open-sdk/crosstool-NG/.build/src/newlib-1.20.0/libgloss/xtensa   ... gcc3
[ALL  ]    checking whether we are using GNU C... yes
[ALL  ]    checking whether xtensa-unknown-elf-cc -B/research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-unknown-elf/build/build-libc/xtensa-unknown-elf/newlib/ -isystem /research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-unknown-elf/build/build-libc/xtensa-unknown-elf/newlib/targ-include -isystem /research/Esp8266/esp-open-sdk/crosstool-NG/.build/src/newlib-1.20.0/newlib/libc/include -B/research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-unknown-elf/build/build-libc/xtensa-unknown-elf/libgloss/xtensa -L/research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-unknown-elf/build/build-libc/xtensa-unknown-elf/libgloss/libnosys -L/research/Esp8266/esp-open-sdk/crosstool-NG/.build/src/newlib-1.20.0/libgloss/xtensa    accepts -g... no
[ALL  ]    checking for xtensa-unknown-elf-as... /research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-unknown-elf/buildtools/lib/gcc/xtensa-unknown-elf/4.8.3/../../../../xtensa-unknown-elf/bin/as
[ALL  ]    checking for xtensa-unknown-elf-ar... /research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-unknown-elf/buildtools/lib/gcc/xtensa-unknown-elf/4.8.3/../../../../xtensa-unknown-elf/bin/ar
[ALL  ]    checking for xtensa-unknown-elf-ranlib... /research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-unknown-elf/buildtools/lib/gcc/xtensa-unknown-elf/4.8.3/../../../../xtensa-unknown-elf/bin/ranlib
[ALL  ]    checking for xtensa-unknown-elf-readelf... xtensa-unknown-elf-readelf
[ALL  ]    checking whether to enable maintainer-specific portions of Makefiles... no
[ALL  ]    *** Newlib does not support CPU xtensa
[ALL  ]    Makefile:6922: recipe for target 'configure-target-newlib' failed
[ERROR]    gmake[2]: *** [configure-target-newlib] Error 1
[ALL  ]    gmake[2]: Leaving directory '/research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-unknown-elf/build/build-libc'
[ALL  ]    Makefile:842: recipe for target 'all' failed
[ERROR]    gmake[1]: *** [all] Error 2
[ALL  ]    gmake[1]: Leaving directory '/research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-unknown-elf/build/build-libc'
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing C library'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_libc[scripts/build/libc/newlib.sh@122]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@638]

jcmvbkbc, 2015-08-04:
@DrObscure can you please share your complete build log?

DrObscure, 2015-08-05:
This message was originally HTML formatted.  View in a HTML capable client to see the original version.\r\n\r\nHi Max.. thanks for the quick response...This morning I realized I had made a rookie mistake and cleaned the crosstool-NG with ./ct-ng clean, not realizing my .config had changed and did not rerun 'configure  xtensa-lx106-elf', causing this failure..  I am using the pfalcon tree and trying to use the gcc 4.9.1 compiler, edited out the line in the makefile after the first run :)As such I apologize for the unwarranted issue posting and will set it as closed after sending you this...However, I still have an issue, possibly another configuration issue...  and hopefully you can give me some pointers here..I have attached the complete gzipped build.log file for your perusal and am including the last few lines for you here..(Would you like for me to create this as another issue on github for historical reference?)--thanks, chane (DrObscure)---- build.log excerpt ----------[ALL  ]    Checking multilib configuration for libgcc...[ALL  ]    mkdir xtensa-lx106-elf[ALL  ]    mkdir xtensa-lx106-elf/libgcc[ALL  ]    Configuring in xtensa-lx106-elf/libgcc[ALL  ]    configure: creating cache ./config.cache[ALL  ]    checking build system type... i386-build_unknown-freebsd10.1[ALL  ]    checking host system type... xtensa-lx106-elf[ALL  ]    checking for --enable-version-specific-runtime-libs... no[ALL  ]    checking for a BSD-compatible install... /research/Esp8266/esp-open-sdk/crosstool-NG/.build/tools/bin/install -c[ALL  ]    checking for gawk... gawk[ALL  ]    checking for xtensa-lx106-elf-ar... /research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/xtensa-lx106-elf/bin/ar[ALL  ]    checking for xtensa-lx106-elf-lipo... xtensa-lx106-elf-lipo[ALL  ]    checking for xtensa-lx106-elf-nm... /research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-core-pass-2/./gcc/nm[ALL  ]    checking for xtensa-lx106-elf-ranlib... /research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/xtensa-lx106-elf/bin/ranlib[ALL  ]    checking for xtensa-lx106-elf-strip... /research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/xtensa-lx106-elf/bin/strip[ALL  ]    checking whether ln -s works... yes[ALL  ]    checking for xtensa-lx106-elf-gcc...  /research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-core-pass-2/./gcc/xgcc -B/research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-core-pass-2/./gcc/ -B/research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/xtensa-lx106-elf/bin/ -B/research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/xtensa-lx106-elf/lib/ -isystem /research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/xtensa-lx106-elf/include -isystem /research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/xtensa-lx106-elf/sys-include[ERROR]    checking for suffix of object files... configure: error: in `/research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-core-pass-2/xtensa-lx106-elf/libgcc':[ERROR]    configure: error: cannot compute suffix of object files: cannot compile[ALL  ]    See `config.log' for more details.[ALL  ]    Makefile:10441: recipe for target 'configure-target-libgcc' failed[ERROR]    gmake[3]: *** [configure-target-libgcc] Error 1[ALL  ]    gmake[3]: Leaving directory '/research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-core-pass-2'[ERROR][ERROR]  [ERROR]    Build failed in step 'Installing pass-2 core C compiler'[ERROR]          called in step '(top-level)'---- Original Message ----From: "Max Filippov" notifications@github.comSent: 8/4/2015 4:30:53 AMTo: "jcmvbkbc/crosstool-NG" crosstool-NG@noreply.github.comCc: "Dr. Obscure" chane@carrollsweb.comSubject: Re: [crosstool-NG] ERROR in build -- [ALL ] *** Newlib does not support CPU xtensa (#14)@DrObscure can you please share your complete build log?

jcmvbkbc, 2015-08-05:
> attached the complete gzipped build.log

It's not here, I guess github strips attachments. Can you share it elsewhere?

> [ERROR] configure: error: cannot compute suffix of object files: cannot compile
> [ALL ] See `config.log' for more details.

Can you also share a file /research/Esp8266/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-core-pass-2/xtensa-lx106-elf/libgcc/config.log ?

DrObscure, 2015-08-05:
Is there a way to attach/upload files to these comments?
I see for images, but not files.. que no?
I can post to Pastebin otherwise..

jcmvbkbc, 2015-08-05:
> Is there a way to attach/upload files to these comments?
> I can post to Pastebin otherwise..

Looks like no. pastebin would be ok.

DrObscure, 2015-08-05:
Well.. got the config.log pasted/posted here: http://pastebin.com/wgAytzjZ
they have a 512K limit, so the 9M build.log won't fit there...:(
If you email me directly, I can send it to you that way perhaps..
don't want to create a github for it, but will it no other choice :)

jcmvbkbc, 2015-08-05:
Ok, config.log was enough:

> /tmp//ccGNLWMl.s: Assembler messages:
> /tmp//ccGNLWMl.s:12: Error: unknown opcode or format name 'entry'
> /tmp//ccGNLWMl.s:16: Error: unknown opcode or format name 'retw.n'

The gcc you've built is not configured properly for call0 ABI, so the code it generates confuses the assembler.

I guess that's for gcc-4.9.1? If so then you need to provide call0 ABI patches to 4.9.1, something like ```ln -s 4.8.2 /research/Esp8266/esp-open-sdk/crosstool-NG/local-patches/gcc/4.9.1```

You can mail me directly to jcmvbkbc@gmail.com

DrObscure, 2015-08-05:
I get a similar error when compiling 4.8.2, which is why I tried 4.9.1
I did notice that the configuration was set to use 4.8.2
I can do another build with that, if you prefer..to keep things at a known point..
I'll send the 4.9.1 build.log anyway...

jcmvbkbc, 2015-08-05:
Ok, thanks for the build.log.

> [EXTRA]    Patching 'binutils-2.24'
> [DEBUG]    Looking for patches in '/research/Esp8266/esp-open-sdk/crosstool-NG/patches/binutils/2.24'...
> [DEBUG]    ==> Executing: 'patch' '--no-backup-if-mismatch' '-g0' '-F1' '-p1' '-f' '-i' '/research/Esp8266/esp-open-sdk/crosstool-NG/patches/binutils/2.24/120-sh-conf.patch'
> [ALL  ]    patch: unrecognized option `--no-backup-if-mismatch'
> [ALL  ]    usage: patch [-bCcEeflNnRstuv] [-B backup-prefix] [-D symbol] [-d directory]
> [ALL  ]                 [-F max-fuzz] [-i patchfile] [-o out-file] [-p strip-count]
> [ALL  ]                 [-r rej-name] [-V t | nil | never] [-x number] [-z backup-ext]
> [ALL  ]                 [--posix] [origfile [patchfile]]
> [ALL  ]           patch <patchfile

Looks like crosstool-NG gives your patch command unrecognized argument. Do you have GNU patch? If not you can try the following patch for the crosstool-NG:

```
diff --git a/scripts/functions b/scripts/functions
index a309f2d..7039e14 100644
--- a/scripts/functions
+++ b/scripts/functions
@@ -1095,7 +1095,7 @@ CT_Patch() {
         if [ -n "${d}" -a -d "${d}" ]; then
             for p in "${d}"/*.patch; do
                 if [ -f "${p}" ]; then
-                    CT_DoExecLog ALL patch --no-backup-if-mismatch -g0 -F1 -p1 -f -i "${p}"
+                    CT_DoExecLog ALL patch -g0 -F1 -p1 -f -i "${p}"
                 fi
             done
             if [ "${CT_PATCH_SINGLE}" = "y" ]; then
```

DrObscure, 2015-08-05:
Thanks Max.. made sure gpatch was installed and in the path now, and added the patch above by hand to 'functions'..will update you when done..

DrObscure, 2015-08-05:
Failed again...  sending you the build.log through the back channel :)

jcmvbkbc, 2015-08-05:
```
[DEBUG]    ==> Executing: 'patch' '-g0' '-F1' '-p1' '-f' '-i' '/research/Esp8266/esp-open-sdk/crosstool-NG/patches/gcc/4.8.2/001_gcc_4_8_2_backport_patches_to_fix_parallel_build_fail_PR_57683.patch'
[ALL  ]    patch: invalid option -- g
[ALL  ]    usage: patch [-bCcEeflNnRstuv] [-B backup-prefix] [-D symbol] [-d directory]
[ALL  ]                 [-F max-fuzz] [-i patchfile] [-o out-file] [-p strip-count]
[ALL  ]                 [-r rej-name] [-V t | nil | never] [-x number] [-z backup-ext]
[ALL  ]                 [--posix] [origfile [patchfile]]
[ALL  ]           patch <patchfile
```
Hmm, the help printed by 'patch' is the same as last time, are you sure it's the right 'patch' utility? Now it chokes on another option.
If it's the right 'patch', can you remove -g0 as well at the same place in scripts/functions? Looks like it should understand all remaining options.

DrObscure, 2015-08-05:
This message was originally HTML formatted.  View in a HTML capable client to see the original version.\r\n\r\nI noticed the continued use of patch also..I have edited the root makefile, changing patch to gpatch, and edited the crosstool-NG makefile to export patch to /usr/local/bin/gpatch...Is there a configuration switch I should use to force the gpatch usage ?I also have patch set to gpatch in my environment..I'll edit the function file to remove the -g option since there is no RCS use in the system now..Thanks again..---- Original Message ----From: "Max Filippov" notifications@github.comSent: 8/5/2015 12:51:27 PMTo: "jcmvbkbc/crosstool-NG" crosstool-NG@noreply.github.comCc: "Dr. Obscure" chane@carrollsweb.comSubject: Re: [crosstool-NG] ERROR in build -- [ALL ] *** Newlib does not support CPU xtensa (#14)[DEBUG] == Executing: 'patch' '-g0' '-F1' '-p1' '-f' '-i' '/research/Esp8266/esp-open-sdk/crosstool-NG/patches/gcc/4.8.2/001_gcc_4_8_2_backport_patches_to_fix_parallel_build_fail_PR_57683.patch'[ALL ] patch: invalid option -- g[ALL ] usage: patch [-bCcEeflNnRstuv] [-B backup-prefix] [-D symbol] [-d directory][ALL ] [-F max-fuzz] [-i patchfile] [-o out-file] [-p strip-count][ALL ] [-r rej-name] [-V t | nil | never] [-x number] [-z backup-ext][ALL ] [--posix] [origfile [patchfile]][ALL ] patch patchfileHmm, the help printed by 'patch' is the same as last time, are you sure it's the right 'patch' utility? Now it chokes on another option.If it's the right 'patch', can you remove -g0 as well at the same place in scripts/functions? Looks like it should understand all remaining options.

DrObscure, 2015-08-05:
This message was originally HTML formatted.  View in a HTML capable client to see the original version.\r\n\r\nI actually changed the line to read 'gpatch -go ...' instead of removing the -g0..I am doing a new build after a clean..update to come---- Original Message ----From: "Max Filippov" notifications@github.comSent: 8/5/2015 12:51:27 PMTo: "jcmvbkbc/crosstool-NG" crosstool-NG@noreply.github.comCc: "Dr. Obscure" chane@carrollsweb.comSubject: Re: [crosstool-NG] ERROR in build -- [ALL ] *** Newlib does not support CPU xtensa (#14)[DEBUG] == Executing: 'patch' '-g0' '-F1' '-p1' '-f' '-i' '/research/Esp8266/esp-open-sdk/crosstool-NG/patches/gcc/4.8.2/001_gcc_4_8_2_backport_patches_to_fix_parallel_build_fail_PR_57683.patch'[ALL ] patch: invalid option -- g[ALL ] usage: patch [-bCcEeflNnRstuv] [-B backup-prefix] [-D symbol] [-d directory][ALL ] [-F max-fuzz] [-i patchfile] [-o out-file] [-p strip-count][ALL ] [-r rej-name] [-V t | nil | never] [-x number] [-z backup-ext][ALL ] [--posix] [origfile [patchfile]][ALL ] patch patchfileHmm, the help printed by 'patch' is the same as last time, are you sure it's the right 'patch' utility? Now it chokes on another option.If it's the right 'patch', can you remove -g0 as well at the same place in scripts/functions? Looks like it should understand all remaining options.

DrObscure, 2015-08-05:
All better now.. got a complete compile for the package from the root directory..
It seems the main issue was getting gpatch propagated throughout the scripts, makefiles, etc..
Thanks for all your help Max..

closed 2015-08-05 by DrObscure