jcmvbkbc/crosstool-NG #15: '.text' will not fit in iram1_0_seg .. link fails on IoT demo example

DrObscure, 2015-08-22:
Max... got the toolchain to build the SDK example IoT demo successfully in FreeBSD 10.1 now,, 
..but am now blowing out the text segment in iram1_0_seg..

I have seen some other discussion on this topic, but not clear if there has been an accepted solution, ie rebuild the libs locally, or not, etc..  

Appreciate any direction here to get the over this last(:-) hurdle ..  
(If this is not the appropriate place to post this, please let me know)

I have posted the output of the compile/link to pastebin here - http://pastebin.com/egnFgqD9

jcmvbkbc, 2015-08-24:
Your libuser.a looks big, almost 20K in the .text section. Try to move most of it to FLASH using ICACHE_FLASH_ATTR or by modifying linker script (like http://www.esp8266.com/viewtopic.php?f=9&t=3949&p=22764#p22772 )

DrObscure, 2015-08-24:
I found that all the functions in the IoT example already have the ICACHE_FLASH_ATTR added..
I have found that what I was missing is the -DICACHE_FLASH needed by the compiler:
%.o: %.c
        $(XT_GCC) -DICACHE_FLASH -mlongcalls $(ILIST) -c $< -o $@
Now it compiles, and links, and appears to be happy.. :)

closed 2015-08-24 by DrObscure