jcmvbkbc/crosstool-NG #16: [ERROR]  configure: error: in `/opt/Espressif/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final'

calamarino, 2015-09-29:
I have tried a lot of different thing but it has been impossible to make it work :( any help is appreciated 
This is my `/opt/Espressif/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/config.log`
```
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by configure, which was
generated by GNU Autoconf 2.64.  Invocation command line was

  $ /opt/Espressif/crosstool-NG/.build/src/gcc-4.8.2/configure --build=x86_64-build_unknown-linux-gnu --host=x86_64-build_unknown-linux-gnu --target=xtensa-lx106-elf --prefix=/opt/Espressif/crosstool-NG/builds/xtensa-lx106-elf --with-local-prefix=/opt/Espressif/crosstool-NG/builds/xtensa-lx106-elf/xtensa-lx106-elf/sysroot --disable-libmudflap --with-sysroot=/opt/Espressif/crosstool-NG/builds/xtensa-lx106-elf/xtensa-lx106-elf/sysroot --with-newlib --enable-threads=no --disable-shared --with-pkgversion=crosstool-NG 1.20.0 --disable-__cxa_atexit --with-gmp=/opt/Espressif/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-mpfr=/opt/Espressif/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-mpc=/opt/Espressif/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-isl=/opt/Espressif/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-cloog=/opt/Espressif/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-libelf=/opt/Espressif/crosstool-NG/.build/xtensa-lx106-elf/buildtools --enable-lto --enable-target-optspace --disable-libgomp --disable-libmudflap --disable-nls --disable-multilib --enable-languages=c

## --------- ##
## Platform. ##
## --------- ##

hostname = yeke
uname -m = x86_64
uname -r = 4.1.0-1-amd64
uname -s = Linux
uname -v = #1 SMP Debian 4.1.3-1 (2015-08-03)

/usr/bin/uname -p = unknown
/bin/uname -X     = unknown

/bin/arch              = unknown
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /opt/Espressif/crosstool-NG/builds/xtensa-lx106-elf/bin
PATH: /opt/Espressif/crosstool-NG/.build/xtensa-lx106-elf/buildtools/bin
PATH: /opt/Espressif/crosstool-NG/.build/tools/bin
PATH: /usr/local/bin
PATH: /usr/bin
PATH: /bin
PATH: /usr/local/games
PATH: /usr/games
PATH: /home/yerco/development/phabricator_tools/arcanist/bin
PATH: /home/yerco/development/phabricator_tools/arcanist/resources/shell
PATH: /opt/debian_armel_wheezy/lib/arm-linux-gnueabi
PATH: /home/yerco/development/esp-open-sdk/xtensa-lx106-elf/bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2317: checking build system type
configure:2331: result: x86_64-build_unknown-linux-gnu
configure:2378: checking host system type
configure:2391: result: x86_64-build_unknown-linux-gnu
configure:2411: checking target system type
configure:2424: result: xtensa-lx106-elf
configure:2478: checking for a BSD-compatible install
configure:2546: result: /opt/Espressif/crosstool-NG/.build/tools/bin/install -c
configure:2557: checking whether ln works
configure:2579: result: yes
configure:2583: checking whether ln -s works
configure:2587: result: yes
configure:2594: checking for a sed that does not truncate output
configure:2658: result: /opt/Espressif/crosstool-NG/.build/tools/bin/sed
configure:2667: checking for gawk
configure:2683: found /usr/bin/gawk
configure:2694: result: gawk
configure:3164: checking for libatomic support
configure:3170: result: no
configure:3183: checking for libitm support
configure:3189: result: no
[... 348 lines cut ...]

## ------------------- ##
## File substitutions. ##
## ------------------- ##

alphaieee_frag=''
host_makefile_frag='/dev/null'
ospace_frag=''
serialization_dependencies=''
target_makefile_frag=''

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME ""
#define PACKAGE_TARNAME ""
#define PACKAGE_VERSION ""
#define PACKAGE_STRING ""
#define PACKAGE_BUGREPORT ""
#define PACKAGE_URL ""

configure: exit 77
```

jcmvbkbc, 2015-09-29:
Looks like your system is missing libstdc++. Do you have it?

calamarino, 2015-09-29:
Thanks for replying. l have `libstdc++-4.9-dev` & `libstdc++6`

jcmvbkbc, 2015-09-30:
> l have libstdc++-4.9-dev & libstdc++6

4.9 doesn't match your compiler version, which is 5.2.
What happens if you do the following:
```
echo 'int main() {return 0;}' > test.c
x86_64-build_unknown-linux-gnu-gcc -O2 -g -pipe -Wl,-t -lstdc++ -lm test.c
```
For me it does this (with x86_64-build_unknown-linux-gnu-gcc changed to x86_64-redhat-linux-gcc as I'm on Fedora):
```
/usr/bin/ld: mode elf_x86_64
/usr/lib/gcc/x86_64-redhat-linux/4.7.2/../../../../lib64/crt1.o
/usr/lib/gcc/x86_64-redhat-linux/4.7.2/../../../../lib64/crti.o
/usr/lib/gcc/x86_64-redhat-linux/4.7.2/crtbegin.o
-lstdc++ (/usr/lib/gcc/x86_64-redhat-linux/4.7.2/libstdc++.so)
-lm (/usr/lib/gcc/x86_64-redhat-linux/4.7.2/../../../../lib64/libm.so)
/tmp/ccHTrfJz.o
-lgcc_s (/usr/lib/gcc/x86_64-redhat-linux/4.7.2/libgcc_s.so)
/lib64/libc.so.6
(/usr/lib64/libc_nonshared.a)elf-init.oS
/lib64/ld-linux-x86-64.so.2
-lgcc_s (/usr/lib/gcc/x86_64-redhat-linux/4.7.2/libgcc_s.so)
/usr/lib/gcc/x86_64-redhat-linux/4.7.2/crtend.o
/usr/lib/gcc/x86_64-redhat-linux/4.7.2/../../../../lib64/crtn.o
```
Note that it gets libstdc++.so in compiler-specific location.

calamarino, 2015-09-30:
Thanks you are genius!

closed 2015-09-30 by calamarino