jcmvbkbc/crosstool-NG #19: Support for later levels of GCC

nkolban, 2015-11-23:
I see that the version of GCC built is 4.8.2.  I see that the latest version of GCC is 5.2 and am wondering if there are plans to update?

jcmvbkbc, 2015-11-23:
It's been updated: the branch lx106-g++-1.21.0 offers gcc-5.1 by default, here's the announcement: http://www.esp8266.com/viewtopic.php?f=9&t=224&start=155#p19771
But it seems that not a lot of people picked it up. So I left 4.8.x as default in xtensa-1.22.x branch, but whoever needs that still can do 'ct-ng menuconfig' after the 'ct-ng xtensa-lx106-elf' and select gcc-5.x.

maruno, 2016-05-16:
Is compiling GCC 5.2 supported? From the xtensa-1.22.x-branch I can't select GCC 5.1, and GCC 5.2 seems to fail to compile here for me: https://gist.github.com/maruno/64e33793c46fc84480169ccd01bbcd10

jcmvbkbc, 2016-05-16:
> unwind-dw2-fde-dip.c:36:40: fatal error: elf.h: No such file or directory

I believe your system is missing libc development package. On ubuntu/debian it should be fixed with
apt-get install build-essential

maruno, 2016-05-18:
I'm not running Linux, but trying to build this on FreeBSD. Shouldn't elf.h be found in the target system headers? Adding the system headers to the last command doesn't seem to work too well.

I updated my gist with the full build.log.

maruno, 2016-05-18:
Adding -isystem /storage/tooling/crosstool-NG/bin/.build/src/gcc-5.2.0/gcc/config/xtensa to the last command seems to let it run, maybe a missing include directory?

jcmvbkbc, 2016-05-18:
Reading [unwind-dw2-fde-dip.c](https://github.com/gcc-mirror/gcc/blob/master/libgcc/unwind-dw2-fde-dip.c) I see the following:
```
#if !defined(inhibit_libc) && !defined(__OpenBSD__)
#include <elf.h>		/* Get DT_CONFIG.  */
#endif
```
which makes me think, that something like "&& !defined(__FreeBSD__)" may be missing here.

jcmvbkbc, 2016-05-18:
OK, reading it a bit further I realize that the compiler thinks that it builds the unwinding library for FreeBSD running on esp8266. At that point I'd say that this is gcc issue and this type of configuration is not actually supported by gcc.

igrr, 2016-10-14:
Getting this issue on macOS as well.
Shouldn't this unwinding code be only added to libgcc build if it is being built for linux target?

jcmvbkbc, 2016-10-16:
Now I'm getting the same error when building esp32 toolchain, but not esp8266. Looking at it.

jcmvbkbc, 2016-10-17:
Ok, my problem was that unwind-dw2-fde-dip.c should not have been used on baremetal elf target. It should be fixed with commit a194053064fba3c261bce32f2eebe16e4d775225. I suspect that it also fixes issues seen on *BSD/macOS, can you guys please check?

igrr, 2016-10-20:
Thanks, this fixes the issue on macOS.