jcmvbkbc/crosstool-NG #24: Error building with gcc 6.1.1 due to incompatible inline declaration in cfns.h

dpgeorge, 2016-05-22:
Building the toolchain using gcc 6.1.1 fails.  The failure is in .build/src/gcc-4.8.2/gcc/cp/cfns.h.  The issue is that the declaration of libc_name_p does not match the definition of this function.  The discrepancy is how the inline attribute is given.  It seems that this only manifests as an error using a recent version of gcc as the host compiler (v5 used to work for me but v6 now fails).

This patch seems to fix the problem:
```
--- cfns.h.orig	2016-05-22 22:59:08.761082973 +0100
+++ cfns.h	2016-05-22 22:49:01.791989920 +0100
@@ -54,6 +54,9 @@
 #ifdef __GNUC__
 __inline
 #endif
+#ifdef __GNUC_STDC_INLINE__
+__attribute__ ((__gnu_inline__))
+#endif
 const char * libc_name_p (const char *, unsigned int);
 /* maximum key range = 391, duplicates = 0 */
```
(Probably cfns.gperf should also be updated in a similar way.)

The issue looks like it was first discovered here: https://gcc.gnu.org/ml/gcc-patches/2015-08/msg00375.html but in a different context.

I don't know if this can/should be fixed in this repo or further upstream?

jcmvbkbc, 2016-05-23:
Hi Damien,
I see the following commit in the mainline that does the same thing according to its changelog: https://github.com/crosstool-ng/crosstool-ng/commit/5a5fcbe148acf35372ecac145816c3515e6b8839
I'll add it to the xtensa-1.22.x branch.

dpgeorge, 2016-05-23:
> I see the following commit in the mainline that does the same thing according to its changelog

Yes that does look correct (but I did not test it).

Right now I am applying my own patch, but having it fixed in this repo would be the best option, thanks!

jcmvbkbc, 2016-05-23:
Fixed with commit ecfc19a597d7

closed 2016-05-23 by jcmvbkbc

dpgeorge, 2016-05-23:
Great, thanks!

ClaymorePT, 2016-06-02:
Could you provide a pull request to the official crosstool-ng stream?
I stumbled upon this issue when compiling gcc 4.9.3 to be used with the Raspbian distro.
It seems that this issue is not fixed upstream.

Thank you.

dpgeorge, 2016-06-02:
@ClaymorePT The patch to fix this comes from upstream crosstool-ng.  They have it applied to gcc 5.3.0 and would just need to apply the same thing to 4.9.3.

ClaymorePT, 2016-06-02:
Ok. Thank you for the info.