jcmvbkbc/crosstool-NG #25: esp108-1.21.0 compile (linker) very slow on OSX

tuanpmt, 2016-06-14:
I've tried on my Mac this command:
`date +"%T" && make && date +"%T"`
This line work very slow, about 4 minutes, 12 seconds:
```
xtensa-esp108-elf-gcc  -L/Users/TuanPM/Projects/2016/espressif/ESP32_RTOS_SDK/lib -nostdlib -T/Users/TuanPM/Projects/2016/espressif/ESP32_RTOS_SDK/ld/pro.map1.1.ld /Users/TuanPM/Projects/2016/espressif/ESP32_RTOS_SDK/ld/pro.rom.addr.ld -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group -lc -lgcc -lhal -lm -lcrypto -lfreertos -llwip -lmain -lnet80211 -lphy -lpp -lrtc -lwpa -lsmartconfig user/.output/eagle/debug/lib/libuser.a sample_lib/.output/eagle/debug/lib/libsample.a -Wl,--end-group -o .output/eagle/debug/image/eagle.app.v7.out
```

while, only need 4 seconds on ubuntu:

[image]

Thanks,

jcmvbkbc, 2016-06-14:
According to your screenshot no compilation or linking took place on ubuntu.
Does anything change if you add -Wl,--no-relax to the linker command line on OSX?

tuanpmt, 2016-06-14:
i'm compile this project, same Makefile: https://github.com/espressif/ESP32_RTOS_SDK/tree/master/examples/project_template

tuanpmt, 2016-06-14:
`-Wl,--no-relax` not work too
```
xtensa-esp108-elf-gcc  -L/Users/TuanPM/Projects/2016/espressif/ESP32_RTOS_SDK/lib -nostdlib -T/Users/TuanPM/Projects/2016/espressif/ESP32_RTOS_SDK/ld/pro.map1.1.ld /Users/TuanPM/Projects/2016/espressif/ESP32_RTOS_SDK/ld/pro.rom.addr.ld -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group -lc -lgcc -lhal -lm -lcrypto -lfreertos -llwip -lmain -lnet80211 -lphy -lpp -lrtc -lwpa -lsmartconfig user/.output/eagle/debug/lib/libuser.a sample_lib/.output/eagle/debug/lib/libsample.a -Wl,--end-group -Wl,--no-relax -o .output/eagle/debug/image/eagle.app.v7.out
```

jcmvbkbc, 2016-06-14:
Ok, then you can try to debug it. Maybe start with running link command under strace -f -c (is there strace in OSX? Does it have -c?) and seeing in which system calls the time is spent.

tuanpmt, 2016-06-14:
i've used dtruss, and result:
[log.txt](https://github.com/jcmvbkbc/crosstool-NG/files/313622/log.txt)

```
................more...............
6414/0x234a5:  open("/Users/TuanPM/Projects/2016/espressif/ESP32_RTOS_SDK/lib/libc.a\0", 0x0, 0x0)		 = 21 0
 6414/0x234a5:  lseek(0x15, 0x7B8E, 0x0)		 = 31630 0
 6414/0x234a5:  read(0x15, "\177ELF\001\001\001\0", 0x10)		 = 16 0
 6414/0x234a5:  lseek(0x15, 0x7B8E, 0x0)		 = 31630 0
 6414/0x234a5:  read(0x15, "\177ELF\001\001\001\0", 0x40)		 = 64 0
 6414/0x234a5:  lseek(0x15, 0x9C5E, 0x0)		 = 40030 0
 6414/0x234a5:  read(0x15, " \0", 0x438)		 = 1080 0
 6414/0x234a5:  lseek(0x15, 0x90DF, 0x0)		 = 37087 0
 6414/0x234a5:  read(0x15, "\0", 0xE1)		 = 225 0
 6414/0x234a5:  lseek(0x15, 0x9C5E, 0x0)		 = 40030 0
 6414/0x234a5:  read(0x15, " \0", 0x438)		 = 1080 0
 6414/0x234a5:  lseek(0x15, 0x90DF, 0x0)		 = 37087 0
 6414/0x234a5:  read(0x15, "\0", 0xE1)		 = 225 0
 6414/0x234a5:  close(0x15)		 = 0 0
 6414/0x234a5:  lseek(0x5, 0x8C000, 0x0)		 = 573440 0
 6414/0x234a5:  read_nocancel(0x5, "\0", 0x1000)		 = 4096 0
 6414/0x234a5:  lseek(0x5, 0x8E000, 0x0)		 = 581632 0
 6414/0x234a5:  read_nocancel(0x5, "\0", 0x1000)		 = 4096 0
 6414/0x234a5:  lseek(0x5, 0x8D000, 0x0)		 = 577536 0
 6414/0x234a5:  read_nocancel(0x5, "R\237\n\0", 0x1000)		 = 4096 0
 6414/0x234a5:  lseek(0x5, 0x8C000, 0x0)		 = 573440 0
 6414/0x234a5:  read_nocancel(0x5, "\0", 0x1000)		 = 4096 0
 6414/0x234a5:  lseek(0x0, 0x0, 0x0)		 = 577536 0
 6413/0x234a3:  wait4(0x190E, 0x100100020, 0x0)		 = 6414 0
 6411/0x23499:  wait4(0x190D, 0x100501BF0, 0x0)		 = 6413 0 
<<== STOP HERE
```

jcmvbkbc, 2016-06-14:
> i've used dtruss, and result

This is an execution trace, it's too detailed and it doesn't have timing information. I was talking about collecting syscall statistics: number of calls and time spent. I see that dtruss supports -c option as well, could you start with that?

tuanpmt, 2016-06-14:
With option -c, additional log:
```
..........
33323/0x64124:  wait4(0x822C, 0x100500760, 0x0)		 = 33324 0
33322/0x64120:  wait4(0x822B, 0x100600010, 0x0)		 = 33323 0

CALL                                        COUNT
execve                                          1
getrusage                                       1
munmap                                          1
dup2                                            2
readlink                                        2
vfork                                           2
wait4                                           2
bsdthread_register                              3
getrlimit                                       3
pread                                           3
proc_info                                       3
shared_region_check_np                          3
statfs64                                        3
getpid                                          4
fcntl_nocancel                                  5
issetugid                                       6
lstat64                                         6
sysctl                                          6
thread_selfid                                   6
mmap                                            7
csops                                           9
close_nocancel                                 12
ioctl                                          14
sigaction                                      22
mprotect                                       24
fstat64                                        26
access                                         27
fcntl                                          37
open_nocancel                                  38
getattrlist                                    77
open                                          147
close                                         154
stat64                                        266
read_nocancel                                 802
read                                          840
lseek                                        1431
```

jcmvbkbc, 2016-06-14:
No time report, unfortunately. And call counts look rather small.
Feel free to try any other profiling method, I'll sure take a look at the results.

tuanpmt, 2016-07-11:
This is solved by: http://esp32.com/viewtopic.php?f=14&t=145&p=731#p669
In `/etc/auto_master`, you need to comment out the line starting with `/home` and then execute:
`sudo automount -vc`