jcmvbkbc/crosstool-NG #26: [ERROR]  Missing: 'x86_64-unknown-linux-gnu-g++' or 'x86_64-unknown-linux-gnu-g++' or 'g++' : either needed!

SharePortal, 2016-06-19:
I tried to install croostool for compiling esp6288 and while i was installing i had error
[ERROR]  Missing: 'x86_64-unknown-linux-gnu-g++' or 'x86_64-unknown-linux-gnu-g++' or 'g++' : either needed!

and i dont know what is meaning i mean i tried to do this structure 
http://docs.kaaproject.org/display/KAA/ESP8266
but i have error 
**could you please help me ??**

`iiot@ubuntu:/opt/Espressif$ git clone -b lx106 git://github.com/jcmvbkbc/crosstool-NG.git
Cloning into 'crosstool-NG'...
remote: Counting objects: 28997, done.
remote: Total 28997 (delta 0), reused 0 (delta 0), pack-reused 28997
Receiving objects: 100% (28997/28997), 15.93 MiB | 1.59 MiB/s, done.
Resolving deltas: 100% (16740/16740), done.
Checking connectivity... done.
iiot@ubuntu:/opt/Espressif$ cd crosstool-NG
iiot@ubuntu:/opt/Espressif/crosstool-NG$ ./bootstrap && ./configure --prefix=`pwd` && make && sudo make install
Running autoconf...
Done. You may now run:
    ./configure
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for a sed that does not truncate output... /bin/sed
checking whether sed understands -r -i -e... yes
checking whether ln -s works... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking how to run the C preprocessor... gcc -E
checking for ranlib... ranlib
checking for gobjcopy... no
checking for objcopy... objcopy
checking for absolute path to objcopy... /usr/bin/objcopy
checking for gobjdump... no
checking for objdump... objdump
checking for absolute path to objdump... /usr/bin/objdump
checking for greadelf... no
checking for readelf... readelf
checking for absolute path to readelf... /usr/bin/readelf
checking for gperf... gperf
checking for absolute path to gperf... /usr/bin/gperf
checking for bison... bison
checking for flex... flex
checking for makeinfo... makeinfo
checking for cut... cut
checking for stat... stat
checking for readlink... readlink
checking for wget... wget
checking for tar... tar
checking for gzip... gzip
checking for bzip2... bzip2
checking for patch... /usr/bin/patch
checking for bash >= 3.1... /bin/bash
checking for GNU awk... /usr/bin/gawk
checking for GNU make >= 3.80... /usr/bin/make
checking whether /usr/bin/make sets $(MAKE)... yes
checking for make 3.81... no
checking for GNU libtool >= 1.5.26... /usr/bin/libtool
checking for GNU libtoolize >= 1.5.26... /usr/bin/libtoolize
checking for GNU automake >= 1.10... /usr/bin/automake
checking for xz... xz
checking for cvs... no
checking for svn... no
checking for inline... inline
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for stdlib.h... (cached) yes
checking for GNU libc compatible malloc... yes
checking for stdlib.h... (cached) yes
checking for GNU libc compatible realloc... yes
checking for size_t... yes
checking for working alloca.h... yes
checking for alloca... yes
checking libintl.h usability... yes
checking libintl.h presence... yes
checking for libintl.h... yes
checking whether gettext is declared... yes
checking ncurses/ncurses.h usability... no
checking ncurses/ncurses.h presence... no
checking for ncurses/ncurses.h... no
checking ncurses/curses.h usability... no
checking ncurses/curses.h presence... no
checking for ncurses/curses.h... no
checking ncursesw/curses.h usability... no
checking ncursesw/curses.h presence... no
checking for ncursesw/curses.h... no
checking ncurses.h usability... yes
checking ncurses.h presence... yes
checking for ncurses.h... yes
checking for library containing initscr... -lncurses
configure: creating ./config.status
config.status: creating Makefile
  SED    'ct-ng'
  SED    'scripts/crosstool-NG.sh'
  SED    'scripts/saveSample.sh'
  SED    'scripts/showTuple.sh'
  GEN    'config/configure.in'
  GEN    'paths.mk'
  GEN    'paths.sh'
  DEP    'nconf.gui.dep'
  DEP    'nconf.dep'
  DEP    'lxdialog/yesno.dep'
  DEP    'lxdialog/util.dep'
  DEP    'lxdialog/textbox.dep'
  DEP    'lxdialog/menubox.dep'
  DEP    'lxdialog/inputbox.dep'
  DEP    'lxdialog/checklist.dep'
  DEP    'mconf.dep'
  DEP    'conf.dep'
  BISON  'zconf.tab.c'
  GPERF  'zconf.hash.c'
  LEX    'lex.zconf.c'
  DEP    'zconf.tab.dep'
  CC     'zconf.tab.o'
  CC     'conf.o'
  LD     'conf'
  CC     'lxdialog/checklist.o'
  CC     'lxdialog/inputbox.o'
  CC     'lxdialog/menubox.o'
  CC     'lxdialog/textbox.o'
  CC     'lxdialog/util.o'
  CC     'lxdialog/yesno.o'
  CC     'mconf.o'
  LD     'mconf'
  CC     'nconf.o'
  CC     'nconf.gui.o'
  LD     'nconf'
  SED    'docs/ct-ng.1'
  GZIP   'docs/ct-ng.1.gz'
  GEN    'config/configure.in'
  GEN    'paths.mk'
  GEN    'paths.sh'
  MKDIR   '/opt/Espressif/crosstool-NG/bin/'
  INST    'ct-ng'
  MKDIR   '/opt/Espressif/crosstool-NG/lib/ct-ng.1.20.0/'
  INSTDIR 'config/'
  INSTDIR 'contrib/'
  INSTDIR 'patches/'
  INSTDIR 'scripts/'
  INST    'steps.mk'
  INST    'paths'
  INSTDIR 'samples/'
  INST    'kconfig/'
  MKDIR   '/opt/Espressif/crosstool-NG/share/doc/crosstool-ng/ct-ng.1.20.0/'
  INST    'docs/*.txt'
  MKDIR   '/opt/Espressif/crosstool-NG/share/man/man1/'
  INST    'ct-ng.1.gz'

For auto-completion, do not forget to install 'ct-ng.comp' into
your bash completion directory (usually /etc/bash_completion.d)
iiot@ubuntu:/opt/Espressif/crosstool-NG$ ./ct-ng xtensa-lx106-elf
  MKDIR config.gen
  IN    config.gen/arch.in
  IN    config.gen/kernel.in
  IN    config.gen/cc.in
  IN    config.gen/binutils.in
  IN    config.gen/libc.in
  IN    config.gen/debug.in
  CONF  config/config.in
#
# configuration saved
#

***********************************************************

Initially reported by: Max Filippov <jcmvbkbc@gmail.com>
URL: http://www.esp8266.com/viewtopic.php?f=9&t=224

***********************************************************

WARNING! This sample may enable experimental features.
         Please be sure to review the configuration prior
         to building and using your toolchain!
Now, you have been warned!

***********************************************************

Now configured for "xtensa-lx106-elf"
iiot@ubuntu:/opt/Espressif/crosstool-NG$ ./ct-ng build
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20160619.043708
[INFO ]  Building environment variables
[ERROR]  Missing: 'x86_64-unknown-linux-gnu-g++' or 'x86_64-unknown-linux-gnu-g++' or 'g++' : either needed!
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_Abort[scripts/functions@331]
[ERROR]  >>        called from: CT_TestAndAbort[scripts/functions@351]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@455]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 0:01.36)
[00:02] / ct-ng:148: recipe for target 'build' failed
make: *** [build] Error 1
`

jcmvbkbc, 2016-06-22:
> ./bootstrap && ./configure --prefix=pwd` && make && sudo make install

sudo is not needed here.

> [ERROR] >> For more info on this error, look at the file: 'build.log'

Can you share the file build.log from the crosstool-NG directory?

SharePortal, 2016-06-22:
Thanks it is working now :-)

closed 2016-06-22 by SharePortal