jcmvbkbc/crosstool-NG #27: Expat 2.1.0 - 404 not found

schmorrison, 2016-07-26:
It seems like Expat 2.1.0 is no longer available at that URI. I modified the file ```config/companion_libs/expat.in``` to replace all instaces of ```2.1.0``` with ```2.2.0```.

This seems to fix the problem I encountered. I can make a formal PR if you would like.

Schmorrison

jcmvbkbc, 2016-07-26:
Just tried
 wget http://downloads.sourceforge.net/project/expat/expat/2.1.0/expat-2.1.0.tar.gz
-- works perfectly for me.

schmorrison, 2016-07-26:
Bizarre, really perplexing.

From the top.

schmorrison, 2016-07-26:
Not sure why I was having a problem before, now I am seeing a different issue with installation of "pass 2 core c compiler". This issue is arising while building a dockerfile that has been working for quite some time.

```[INFO ]  Installing pass-2 core C gcc compiler
[ERROR]    collect2: error: ld terminated with signal 9 [Killed]
[ERROR]    make[4]: *** [lto1] Error 1
[ERROR]    make[3]: *** [all-gcc] Error 2
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing pass-2 core C gcc compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_gcc_core_backend[scripts/build/cc/100-gcc.sh@545]
[ERROR]  >>        called from: do_gcc_core_pass_2[scripts/build/cc/100-gcc.sh@160]
[ERROR]  >>        called from: do_cc_core_pass_2[scripts/build/cc.sh@42]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-55-gecfc19a/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 20:52.87)
[20:53] / make[2]: *** [build] Error 2
make[1]: *** [_toolchain] Error 2
make: *** [/home/espbuilder/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2```

schmorrison, 2016-07-26:
Thanks a billion. Building a docker image on a digitalocean droplet. Saw that line and disregarded it for the more important looking lines.

schmorrison, 2016-07-26:
Word, consider this closed. :+1:

closed 2016-07-26 by schmorrison