jcmvbkbc/crosstool-NG #28: Newlib does not support CPU xtensa

jackwilsdon, 2016-08-11:
I'm experiencing an issue with compiling xtensa-lx106-elf with newlib as the standard library, as I am receiving the message "Newlib does not support CPU xtensa" whenever I run `ct-ng build`.

## Files

 - [`.config`](https://gist.github.com/anonymous/4b4b9bbc20125e346c8d98fc3d91323c)
 - [`build.log`](https://gist.github.com/anonymous/70b579f229e810ef7870fbe1d7f1422c) (lag warning - large file)
 - [`/tmp/xtensa-build/xtensa-lx106-elf/build/build-libc/config.log`](https://gist.github.com/anonymous/544a54dd634ffce06023a864045e31f5)
 - [`/tmp/xtensa-build/xtensa-lx106-elf/build/build-libc/xtensa-lx106-elf/newlib/config.log`](https://gist.github.com/anonymous/99e3b3bdb25c2d26234165df603dbb2b)

If you need any more information please don't hesitate to ask.

jrobeson, 2016-10-16:
possible with this fork? https://github.com/espressif/crosstool-NG

creatica-soft, 2020-07-02:
Hm... I am using espressif fork and getting the same error.

jcmvbkbc, 2020-07-02:
@creatica-soft you can enable experimental features and then specify newlib repository by URL. Chosen newlib version doesn't matter in that case.
See [this config](https://github.com/jcmvbkbc/crosstool-NG/blob/xtensa-1.24.x/samples/aarch64-linux-gnu%2Cxtensa-lx106-elf/crosstool.config) for example.