jcmvbkbc/crosstool-NG #32: Error building crosstool-NG on FreeBSD 10.3

timlag1305, 2016-09-26:
I tried building crosstool-NG from source as part of the build process for this SDK: https://github.com/pfalcon/esp-open-sdk

It starts building `gcc-4.8.5` but an error occurs with `ld`
I made sure to install the gnu version of the userland tools including gmake, gpatch, binutils, wget

Here is a link to `build.log`: https://psu.box.com/s/8yx1r3euxedtme8jat8bcelke1hcyseo

jcmvbkbc, 2016-09-26:
Hello,

> [ALL  ]    x86_64-build_unknown-freebsd10.3-g++   -O2 -g -pipe  -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE  -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc -lstdc++ -lm -o lto1 	lto/lto-lang.o lto/lto.o lto/lto-object.o attribs.o lto/lto-partition.o libbackend.a main.o  libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a -L/usr/home/tim/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/lib -lcloog-isl -L/usr/home/tim/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/lib -lisl -L/usr/home/tim/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/lib -L/usr/home/tim/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/lib -L/usr/home/tim/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/lib -lmpc -lmpfr -lgmp -rdynamic  -L../zlib -lz 
> libcommon.a ../libcpp/libcpp.a  -liconv ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a 
> [ALL  ]    /usr/local/bin/ld: BFD (GNU Binutils) 2.27 internal error, aborting at merge.c:909 in bfd_vma _bfd_merged_section_offset(bfd *, asection **, void *, bfd_vma)

That's the host linker failure. You can report it directly to the binutils mailing list at binutils@sourceware.org
FWIW it builds well for me on linux with binutils-2.25.

timlag1305, 2016-09-26:
Okay, thanks for the insight!  I will be sure to send a bug report to that mailing list.

closed 2016-09-26 by timlag1305