jcmvbkbc/crosstool-NG #35: ESP32 Build fails in libgcc on -mlittle-endian compiler switch

DrObscure, 2017-01-18:
Building crosstool-NG on FreeBSD 11.0 for the ESP32.  GCC = 5.2.0
When running the original .config with no declaration on endianness, I got errors due to not claiming one, so added CT_ARCH_ENDIAN_CFLAG="-mlittle-endian" to my .config. (The esp32 docs show it to be little endian in operation)

Now I get an error on Pass-2 of the compiler build as follows:
configure:3464:  /wrk/Espressif/crosstool-NG-xtensa-1.22.x/.build/xtensa-esp32-elf/build/build-cc-gcc-core-pass-2/./gcc/xgcc -B/wrk/Espressif/crosstool-NG-xtensa-1.22.x/.build/xtensa-esp32-elf/build/build-cc-gcc-core-pass-2/./gcc/ -B/wrk/Espressif/crosstool-NG-xtensa-1.22.x/.build/xtensa-esp32-elf/buildtools/xtensa-esp32-elf/bin/ -B/wrk/Espressif/crosstool-NG-xtensa-1.22.x/.build/xtensa-esp32-elf/buildtools/xtensa-esp32-elf/lib/ -isystem /wrk/Espressif/crosstool-NG-xtensa-1.22.x/.build/xtensa-esp32-elf/buildtools/xtensa-esp32-elf/include -isystem /wrk/Espressif/crosstool-NG-xtensa-1.22.x/.build/xtensa-esp32-elf/buildtools/xtensa-esp32-elf/sys-include    -o conftest -mlittle-endian       -mlongcalls -g -Os   conftest.c  >&5
xgcc: error: unrecognized command line option '-mlittle-endian'
configure:3467: $? = 1
configure:3655: checking for suffix of object files
configure:3677:  /wrk/Espressif/crosstool-NG-xtensa-1.22.x/.build/xtensa-esp32-elf/build/build-cc-gcc-core-pass-2/./gcc/xgcc -B/wrk/Espressif/crosstool-NG-xtensa-1.22.x/.build/xtensa-esp32-elf/build/build-cc-gcc-core-pass-2/./gcc/ -B/wrk/Espressif/crosstool-NG-xtensa-1.22.x/.build/xtensa-esp32-elf/buildtools/xtensa-esp32-elf/bin/ -B/wrk/Espressif/crosstool-NG-xtensa-1.22.x/.build/xtensa-esp32-elf/buildtools/xtensa-esp32-elf/lib/ -isystem /wrk/Espressif/crosstool-NG-xtensa-1.22.x/.build/xtensa-esp32-elf/buildtools/xtensa-esp32-elf/include -isystem /wrk/Espressif/crosstool-NG-xtensa-1.22.x/.build/xtensa-esp32-elf/buildtools/xtensa-esp32-elf/sys-include    -c -mlittle-endian       -mlongcalls -g -Os  conftest.c >&5
xgcc: error: unrecognized command line option '-mlittle-endian'
configure:3681: $? = 1
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "GNU C Runtime Library"
| #define PACKAGE_TARNAME "libgcc"
| #define PACKAGE_VERSION "1.0"
| #define PACKAGE_STRING "GNU C Runtime Library 1.0"
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL "http://www.gnu.org/software/libgcc/"
| /* end confdefs.h.  */
|
| int
| main ()
| {
|
|   ;
|   return 0;
| }
configure:3695: error: in `/wrk/Espressif/crosstool-NG-xtensa-1.22.x/.build/xtensa-esp32-elf/build/build-cc-gcc-core-pass-2/xtensa-esp32-elf/libgcc':
configure:3698: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details.
-------------------
How does one rectify this conundrum :) 
Did I do the configuration incorrectly?

jcmvbkbc, 2017-01-18:
> Building crosstool-NG on FreeBSD 11.0 for the ESP32. GCC = 5.2.0
> When running the original .config with no declaration on endianness, I got errors due to not claiming one

Can you please provide the full build log for the initial failing build? I've checked that xtensa-esp32-elf config with gcc-5.2.0 builds well for me on linux.

> so added CT_ARCH_ENDIAN_CFLAG=-mlittle-endian to my .config

This flag is not supported by xtensa gcc.

Here's the mainline fix for the endianness issue I had building uclibc-ng: https://github.com/crosstool-ng/crosstool-ng/commit/6c6829a0610e6282be922183c9601cb2365df9a4
It may be related to what you see.

DrObscure, 2017-01-18:
Thanks...
I implemented the changes as suggested above and, unfortunately now I am back to the fail because of not declaring the endianness :) 

I have included the build.log for your perusal..
-thx again..
[build.log.gz](https://github.com/jcmvbkbc/crosstool-NG/files/715264/build.log.gz)

jcmvbkbc, 2017-01-18:
> I have included the build.log

You have newlib-2.1.0 selected in your config, but there's only patches for newlib-2.0.0 with xtensa support in the branch xtensa-1.22.x. That's the reason.

DrObscure, 2017-01-18:
Great.. that will be an easy cure.. I'll reconfigure and restart the build.. thanks again

DrObscure, 2017-01-21:
one last gotcha, the 'strip' of the binaries failed due to the '-v' flag.. 
must be a change in the newer release.. made a change to internals.sh taking it out, and all went as expected to completion..  
again thanks for all your efforts on this project..