jcmvbkbc/crosstool-NG #39: Issue during make

Aterocana, 2017-03-04:
After executing `./bootstrap` and `./configure`, when I ran `make` I've got this error:

``
In file included from zconf.tab.c:213:0:
zconf.hash.c:163:1: error: conflicting types for ‘kconf_id_lookup’
 kconf_id_lookup (register const char *str, register size_t len)
 ^~~~~~~~~~~~~~~
zconf.hash.c:34:25: note: previous declaration of ‘kconf_id_lookup’ was here
 static struct kconf_id *kconf_id_lookup(register const char *str, register unsigned int len);
                         ^~~~~~~~~~~~~~~
make[2]: *** [Makefile:96: zconf.tab.o] Error 1
make[1]: *** [Makefile:166: build-lib-kconfig] Error 2
make: *** [Makefile:126: build] Error 2
``

I'm in the branch lx106-g++

jcmvbkbc, 2017-03-04:
That's a too recent version of gperf. Should be fixed in lx106-g++ with commit 2fd0c00.

Aterocana, 2017-03-05:
Sorry, I don't get it. Should it be fixed? Then how come I can't compile it?
I try to downgrade my gperf to versions listed here: https://archive.archlinux.org/packages/g/gperf/
Still the error is there.  Thanks for the help.

jcmvbkbc, 2017-03-05:
> Should it be fixed?

Yes, I've just pushed the fix. You need to pull from the branch lx106-g++ to get it.

> I try to downgrade my gperf to versions listed here: https://archive.archlinux.org/packages/g/gperf/
> Still the error is there.

It definitely works with gperf-3.0.4

Aterocana, 2017-03-06:
Ok, thank you. Somehow I had to restart the computer in order to compile it.

closed 2017-03-06 by Aterocana