jcmvbkbc/crosstool-NG #43: Does not compile on Bash for Windows 10

skbobade, 2017-06-03:
In new version of WSL, Ubuntu 16.04.1 LTS, making toolchain fail in step 'Installing ISL for host'

Logs:
Makefile:181: warning: overriding recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir' Makefile:176: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir' make -C crosstool-NG -f ../Makefile _ct-ng make[1]: Entering directory '/mnt/d/ESP82/esp-open-sdk/crosstool-NG' ../Makefile:181: warning: overriding recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir' ../Makefile:176: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir' ./bootstrap Running autoconf... Done. You may now run: ./configure ./configure --prefix=pwd`
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for a sed that does not truncate output... /bin/sed
checking whether sed understands -r -i -e... yes
checking whether ln -s works... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking how to run the C preprocessor... gcc -E
checking for ranlib... ranlib
checking for gobjcopy... no
checking for objcopy... objcopy
checking for absolute path to objcopy... /usr/bin/objcopy
checking for gobjdump... no
checking for objdump... objdump
checking for absolute path to objdump... /usr/bin/objdump
checking for greadelf... no
checking for readelf... readelf
checking for absolute path to readelf... /usr/bin/readelf
checking for gperf... gperf
checking for absolute path to gperf... /usr/bin/gperf
checking for bison... bison
checking for flex... flex
checking for makeinfo... makeinfo
checking for cut... cut
checking for stat... stat
checking for readlink... readlink
checking for wget... wget
checking for tar... tar
checking for gzip... gzip
checking for bzip2... bzip2
checking for help2man... help2man
checking for gpatch... no
checking for patch... patch
checking for absolute path to patch... /usr/bin/patch
checking for bash >= 3.1... /bin/bash
checking for GNU awk... /usr/bin/gawk
checking for GNU make >= 3.80... /usr/bin/make
checking whether /usr/bin/make sets $(MAKE)... yes
checking for make 3.81... no
checking for GNU libtool >= 1.5.26... /usr/bin/libtool
checking for GNU libtoolize >= 1.5.26... /usr/bin/libtoolize
checking for GNU automake >= 1.10... /usr/bin/automake
checking for xz... xz
checking for cvs... no
checking for svn... no
checking for inline... inline
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
[... 141 lines cut ...]
[ERROR] make[3]: *** [/mnt/d/ESP82/esp-open-sdk/crosstool-NG/.build/src/isl-0.14/Makefile.in] Error 1
[ERROR]
[ERROR] >>
[ERROR] >> Build failed in step 'Installing ISL for host'
[ERROR] >> called in step '(top-level)'
[ERROR] >>
[ERROR] >> Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR] >> called from: do_isl_backend[scripts/build/companion_libs/121-isl.sh@125]
[ERROR] >> called from: do_isl_for_host[scripts/build/companion_libs/121-isl.sh@73]
[ERROR] >> called from: do_companion_libs_for_host[scripts/build/companion_libs.sh@36]
[ERROR] >> called from: main[scripts/crosstool-NG.sh@646]
[ERROR] >>
[ERROR] >> For more info on this error, look at the file: 'build.log'
[ERROR] >> There is a list of known issues, some with workarounds, in:
[ERROR] >> 'share/doc/crosstool-ng/crosstool-ng-1.22.0-55-gecfc19a/B - Known issues.txt'
[ERROR]
[ERROR] (elapsed: 19:03.90)
[19:06] / ct-ng:152: recipe for target 'build' failed
make[2]: *** [build] Error 2
make[2]: Leaving directory '/mnt/d/ESP82/esp-open-sdk/crosstool-NG'
../Makefile:130: recipe for target '_toolchain' failed
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory '/mnt/d/ESP82/esp-open-sdk/crosstool-NG'
Makefile:126: recipe for target '/mnt/d/ESP82/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc' failed
make: *** [/mnt/d/ESP82/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2`

jcmvbkbc, 2017-06-03:
Please share the build.log from the crosstool-NG directory.

skbobade, 2017-06-03:
Attached build.log, config.log and, .config files in zip as I ran make again in a fresh installation.

[crosstool-NG.zip](https://github.com/jcmvbkbc/crosstool-NG/files/1049580/crosstool-NG.zip)

jcmvbkbc, 2017-06-03:
> [ALL  ]    /mnt/d/Embedded/esp-open-sdk/crosstool-NG/.build/src/isl-0.14/missing: line 81: automake-1.14: command not found
> [ALL  ]    WARNING: 'automake-1.14' is missing on your system.

I'd say 'just do the obvious'...

skbobade, 2017-06-03:
Automake 1.15 is installed and recognized. Is version 1.14 a requirement? 

`automake --version`
`automake (GNU automake) 1.15`
`Copyright (C) 2014 Free Software Foundation, Inc.`

skbobade, 2017-06-04:
Installed automake-1.14 from source. That eliminated the error.
Ubuntu 16.04 does not have automake-1.14 in apt repository. 
Request including support for newer versions of automake.

jcmvbkbc, 2017-06-04:
> Automake 1.15 is installed and recognized. Is version 1.14 a requirement?

Well, AFAIK automake shouldn't be invoked at all unless parts of the project build system are changed. It's definitely not called for ISL in the builds that I run.

skbobade, 2017-06-04:
I was doing a fresh build of the toolchain itself.