jcmvbkbc/crosstool-NG #47: TUI makes problems under windows cygwin in cross-gdb espressif's [future/psram_malloc] branch

ESP32DE, 2017-07-23:
Hi Max
i [try to build for ESP32](https://esp32.com/viewtopic.php?f=13&t=2031&p=11902#p11887) a [future/psram_malloc branch](https://github.com/espressif/crosstool-NG/tree/xtensa-1.22.x-ext_ram_patch) on espressif fork for ESP32.

cause i can't create an report for Issues there ( not possible to create on the fork )
i hope you or @igrr or other user can say, what is going wrong with the final step cross-debug and TUI

FYI: i have successfull build this toolchain under linux.
now i try to build under windows with cygwin for the windows user cause [there is no officially windows support for this toolchain](https://esp32.com/viewtopic.php?f=13&t=2031#p9568) 

the windows cygwin build process is pretty fine - but at the end with cross-gdb
the build is broken and stop.

a total log is here
[build.zip](https://github.com/jcmvbkbc/crosstool-NG/files/1168453/build.zip)

the build process is run pretty fine in the 95 prozent

```
$ CT_DEBUG_CT_SAVE_STEPS=1
$ ./ct-ng build
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20170723.200623
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[INFO ]  Retrieving needed toolchain components' tarballs: done in 450.18s (at 07:57)
[INFO ]  =================================================================
[INFO ]  Extracting and patching toolchain components
[INFO ]  Extracting and patching toolchain components: done in 239.52s (at 11:57)
[INFO ]  =================================================================
[INFO ]  Installing ncurses for build
[INFO ]  Installing ncurses for build: done in 250.05s (at 16:07)
[INFO ]  =================================================================
[INFO ]  Installing GMP for host
[INFO ]  Installing GMP for host: done in 391.75s (at 22:39)
[INFO ]  =================================================================
[INFO ]  Installing MPFR for host
[INFO ]  Installing MPFR for host: done in 344.25s (at 28:24)
[INFO ]  =================================================================
[INFO ]  Installing ISL for host
[INFO ]  Installing ISL for host: done in 217.70s (at 32:02)
[INFO ]  =================================================================
[INFO ]  Installing MPC for host
[INFO ]  Installing MPC for host: done in 143.63s (at 34:25)
[INFO ]  =================================================================
[INFO ]  Installing expat for host
[INFO ]  Installing expat for host: done in 70.53s (at 35:36)
[INFO ]  =================================================================
[INFO ]  Installing binutils for host
[INFO ]  Installing binutils for host: done in 616.61s (at 45:53)
[INFO ]  =================================================================
[INFO ]  Installing C library headers & start files
[INFO ]  Installing C library headers & start files: done in 0.93s (at 45:54)
[INFO ]  =================================================================
[INFO ]  Installing pass-2 core C gcc compiler
[INFO ]  Installing pass-2 core C gcc compiler: done in 1566.34s (at 72:01)
[INFO ]  =================================================================
[INFO ]  Installing C library
[INFO ]  Installing C library: done in 383.47s (at 78:24)
[INFO ]  =================================================================
[INFO ]  Installing final gcc compiler
[INFO ]  Installing final gcc compiler: done in 2131.73s (at 113:56)
[INFO ]  =================================================================
[INFO ]  Installing cross-gdb
[ERROR]    collect2: error: ld returned 1 exit status
[ERROR]    make[3]: *** [gdb.exe] Error 1
[ERROR]    make[2]: *** [all-gdb] Error 2
[ERROR]    make[1]: *** [all] Error 2
[ERROR]  -
[ERROR]  >>
[... 56 lines cut ...]
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_debug_gdb_build[scripts/build/debug/300-gdb.sh@120]
[ERROR]  >>        called from: do_debug[scripts/build/debug.sh@35]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'docs/B - Known issues.txt'
[ERROR]  
[ERROR]  (elapsed: 133:44.83)

```

i am not sure, what is here the mistake, cause the "master toolchain" for esp32 runs with windows and cygwin, only this psram_patch do not successfull run until end.

perhabs you can look up one time in the log and see the mistake?

just in time i try [to disable the tui for gdb](https://esp32.com/viewtopic.php?f=13&t=2031&p=11902#p11902)  the build process runs just in time. but i think, this will also be broken in the endphase in the scriptings.

thank you for any info or help or further steps

best wishes
rudi ;-)

jcmvbkbc, 2017-07-23:
> cause i can't create an report for Issues there ( not possible to create on the fork )

This repository is also a fork.

ESP32DE, 2017-07-23:
@jcmvbkbc 
txs for your reply, 

> This repository is also a fork.

but i can create "functionally" an Issues report 
but not on espressif fork, there is no button for this.

[image]

this means, i must do open the Issues report [on the main repo foss-xtensa](https://github.com/foss-xtensa/crosstool-NG) from last edit 2013 ? 

i do not really believe this helps - cause the last activ was Nov 11, 2013 - but ok :)

i will try after no successfull with the try disable tui by --disable-tui

thank you for your help.

sorry for the report. 

best wishes
rudi ;-)

closed 2017-07-23 by ESP32DE

jcmvbkbc, 2017-07-23:
> this means, i must do open the Issues report on the main repo foss-xtensa from last edit 2013 ?

No, I guess that espressif need to enable issue tracker on their fork, see my (multiple times) edited reply.

ESP32DE, 2017-07-23:
@jcmvbkbc 
yes - this is deactive Max. ok - no problem!
i will try to animate espressif to activate this issue tracker.
thank you Max! you have helped in this way further ( now i know how "the thing" is named)
best wishes
rudi ;-)
 ( thank you for your efforts in your work for us ( esp8266, esp32.... )