jcmvbkbc/crosstool-NG #48: error: format not a string literal and no format arguments

dotlambda, 2017-08-09:
I'm trying to compile crosstool-NG on NixOS (64-bit Linux), but I'm getting the following error during the step "Installing pass-2 core C gcc compiler" (from `build.log`):
```
[ALL  ]    gcc -c -DHAVE_CONFIG_H -O2 -g -pipe -Wno-error=format-security  -I. -I/home/rschuetz/dev/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/libiberty/../include  -W -Wall -Wwrite-strings -Wc++-compat -Wstrict-prototypes -pedantic  /home/rschuetz/dev/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/libiberty/floatformat.c -o floatformat.o
[ALL  ]    /home/rschuetz/dev/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/libcpp/expr.c: In function 'unsigned int cpp_classify_number(cpp_reader*, const cpp_token*, const char**, source_location)':
[ERROR]    /home/rschuetz/dev/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/libcpp/expr.c:629:18: error: format not a string literal and no format arguments [-Werror=format-security]
[ALL  ]            0, message);
[ALL  ]                      ^
[ERROR]    /home/rschuetz/dev/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/libcpp/expr.c:632:39: error: format not a string literal and no format arguments [-Werror=format-security]
[ALL  ]               virtual_location, 0, message);
[ALL  ]                                           ^
```
They even do not disappear when I add
```
CT_EXTRA_CFLAGS_FOR_BUILD="-Wno-error=format-security"
CT_EXTRA_CFLAGS_FOR_HOST="-Wno-error=format-security"
CT_TARGET_CFLAGS="-Wno-error=format-security"
```
to `.config`.

Do you have an idea how to solve this?

dotlambda, 2017-08-29:
ping

dotlambda, 2017-09-18:
You can also have a look at my [build.log](https://gist.github.com/dotlambda/1fb0cffd187c09415935dc60994ac964) if you want to.

jcmvbkbc, 2017-09-18:
The option -Wno-error was not passed to the compiler in your build:

> [ALL  ]    g++  -I/home/rschuetz/dev/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/libcpp -I. -I/home/rschuetz/dev/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/libcpp/../include -I/home/rschuetz/dev/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/libcpp/include  -O2 -g -pipe  -W -Wall -Wno-narrowing -Wwrite-strings -Wmissing-format-attribute -pedantic -Wno-long-long  -fno-exceptions -fno-rtti -I/home/rschuetz/dev/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/libcpp -I. -I/home/rschuetz/dev/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/libcpp/../include -I/home/rschuetz/dev/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/libcpp/include  -c -o expr.o -MT expr.o -MMD -MP -MF .deps/expr.Tpo /home/rschuetz/dev/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/libcpp/expr.c

And in general it is absent in your build log:

> [DEBUG]    CT_EXTRA_CFLAGS_FOR_HOST=""

Putting it into the CT_EXTRA_CFLAGS_FOR_HOST sounds right, I've checked locally that in the branch xtensa-1.22.x it does what's expected.

dotlambda, 2017-09-18:
I'm sorry, I uploaded a `build.log` without the option.
[Here](https://gist.github.com/dotlambda/3bfe69e9f54f429f853890bc772e987f) you can find one with the option `-Wno-error=format-security` enabled. As stated in the very first message in this issue, it produces the same output.
I really have now idea why this doesn't suppress the error.

jcmvbkbc, 2017-09-18:
Aha, so it fails even when the option is passed. Ok. I see one thing in your build that is different in mine: the name of the host compiler. It's g++ in your case, but x86_64-build_pc-linux-gnu-g++ in mine. In my case it's actually a script that ct-ng put into .build/xtensa-unknown-elf/buildtools/bin. In your case I also see the wrapper script created that points to /nix/store/jdpsfawnxcn76f50387lpsvqmpwccz2l-gcc-wrapper-6.4.0/bin/g++, which looks like another wrapper. Can you start with checking that your g++ (which I suspect is actually /nix/store/jdpsfawnxcn76f50387lpsvqmpwccz2l-gcc-wrapper-6.4.0/bin/g++) does not add any fancy flags into the actual compiler command line?

dotlambda, 2017-10-15:
You're absolutely right. It's right there in the documentation:
https://nixos.org/nixpkgs/manual/#sec-hardening-in-nixpkgs
It's as simple as `export hardeningDisable="format".

closed 2017-10-15 by dotlambda

vyorkin, 2020-04-08:
@dotlambda Could you please post your shell.nix?