jcmvbkbc/crosstool-NG #50: Problems with corrupt binutils tarball on Debian

grodansparadis, 2017-10-23:
I try to compile on debian jessie and always fail on the binutils unpack stage, l

patching file progs/tset.c
[DEBUG]    Looking for patches in '/home/esp/crosstool-NG/lib/crosstool-ng-1.22.0-73-ge28a011/patches/ncurses/6.0/xtensa'...
[DEBUG]    Looking for patches in '/home/esp/crosstool-NG/local-patches/ncurses/6.0'...
[ALL  ]    Overiding config.guess and config.sub
[ALL  ]    mode of './config.guess' retained as 0755 (rwxr-xr-x)
[ALL  ]    '/home/esp/crosstool-NG/scripts/config.guess' -> './config.guess'
[ALL  ]    mode of './config.sub' retained as 0755 (rwxr-xr-x)
[ALL  ]    '/home/esp/crosstool-NG/scripts/config.sub' -> './config.sub'
[DEBUG]    ==> Executing: 'touch' '/home/esp/crosstool-NG/.build/src/.ncurses-6.0.patched'
[DEBUG]    ==> Executing: 'rm' '-f' '/home/esp/crosstool-NG/.build/src/.ncurses-6.0.patching'
[DEBUG]    ==> Executing: 'touch' '/home/esp/crosstool-NG/.build/src/.binutils-2.25.1.extracting'
[DEBUG]    Entering '/home/esp/crosstool-NG/.build/src'
[EXTRA]    Extracting 'binutils-2.25.1'
[DEBUG]    ==> Executing: 'mkdir' '-p' 'binutils-2.25.1'

Checking the tarball it is actually corrupt and can't be unpacked. The rest of the tarballs in   .build/tarballs/ appears to unpack OK also manually

Any workaround or hint to this?

jcmvbkbc, 2017-10-23:
> Checking the tarball it is actually corrupt and can't be unpacked. The rest of the tarballs in .build/tarballs/ appears to unpack OK also manually
> 
> Any workaround or hint to this?

Remove the tarball and rerun the build?

grodansparadis, 2017-10-23:
Have done that a couple of times with the same result.

grodansparadis, 2017-10-23:
Copying in binutils manually and removing the partly extracted stuff makes it unpack. But now it stops at the patching instead.

==> Executing: 'touch' '/home/esp/crosstool-NG/.build/src/.binutils-2.24.extracted'
[DEBUG]    ==> Executing: 'rm' '-f' '/home/esp/crosstool-NG/.build/src/.binutils-2.24.extracting'
[DEBUG]    Entering '/home/esp/crosstool-NG/.build/src/binutils-2.24'
[EXTRA]    Patching 'binutils-2.24'
[DEBUG]    Looking for patches in '/home/esp/crosstool-NG/lib/ct-ng.1.20.0/patches/binutils/2.24'...
[DEBUG]    ==> Executing: 'patch' '--no-backup-if-mismatch' '-g0' '-F1' '-p1' '-f' '-i' '/home/esp/crosstool-NG/lib/ct-ng.1.20.0/patches/binutils/2.24/120-sh-conf.patch'
[ALL  ]    patching file configure
[ALL  ]    Hunk #1 FAILED at 3570.
[ALL  ]    1 out of 1 hunk FAILED -- saving rejects to file configure.rej
[ALL  ]    patching file configure.ac
[ALL  ]    Hunk #1 FAILED at 1006.
[ALL  ]    1 out of 1 hunk FAILED -- saving rejects to file configure.ac.rej
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Extracting and patching toolchain components'
[ERROR]  >>        called in step '(top-level)'

grodansparadis, 2017-10-23:
Yes copying [http://ftp.gnu.org/gnu/binutils/binutils-2.24.tar.gz](http://ftp.gnu.org/gnu/binutils/binutils-2.24.tar.gz) solve the problem.  

So if someone else is getting this error

- remove the partly extracted binutils  -  `rm -rf  .build/src/binutils*`
- remove flags -  `rm .build/src/ .binutils-*`
- remove the bad tarball - `rm .buils/tarballs/binutils*`
- copy the tarball from the gnu link to ./builds/tarballs
- Continue with the build - `./ct-ng build`

Thanks @jcmvbkbc  for your and prompt and kind help! Very much appreciated.

closed 2017-10-23 by grodansparadis