jcmvbkbc/crosstool-NG #51: Support ASAN (-fsanitize=address) for xtensa target

paulreimer, 2017-12-02:
Not sure what is involved, but I know that ASAN support on x86 is amazing for finding tricky bugs, and I'd love to run it on my ESP32 debug builds.

I'm not sure what is involved in adding it for a new target. The current state seems to be a compiler warning (after adding `-fsanitize=address` to `CXXFLAGS`):
```
warning: -fsanitize=address not supported for this target
```

jcmvbkbc, 2017-12-02:
It does not belong to the crosstool-ng, the gcc and libsanitizer bundled with it need to be modified so please move it to xtensa-gcc.
Also compiling with ASAN results in higher memory requirements: according to https://github.com/google/sanitizers/wiki/AddressSanitizerAlgorithm it requires 1/8 of used memory for its shadow map.

closed 2017-12-02 by jcmvbkbc

paulreimer, 2017-12-02:
Thanks @jcmvbkbc, I have added it here:
https://github.com/jcmvbkbc/gcc-xtensa/issues/9