jcmvbkbc/crosstool-NG #67: Q: how to compile toolchain that runs on Android?

4ntoine, 2020-06-12:
Hey, guys. I'm trying to compile a toolchain that is build on x86_64, hosts on ARM (or x86_64) Android and compiles per xtensa lx106. Any suggestions on how can i do it or start at least?

4ntoine, 2020-06-12:
as i can see it's not merged to upstream crosstool-ng (installed with brew):
```
> ct-ng list-samples
Status  Sample name
[G...]   aarch64-rpi3-linux-gnu
[G..X]   aarch64-unknown-linux-android
[G...]   aarch64-unknown-linux-gnu
[G...]   aarch64-unknown-linux-uclibc
[G...]   alphaev56-unknown-linux-gnu
[G...]   alphaev67-unknown-linux-gnu
[G...]   arc-arc700-linux-uclibc
[G...]   arc-multilib-elf32
[G...]   arc-multilib-linux-uclibc
[G...]   arm-bare_newlib_cortex_m3_nommu-eabi
[G...]   arm-cortex_a15-linux-gnueabihf
[G...]   arm-cortex_a8-linux-gnueabi
[G..X]   arm-cortexa5-linux-uclibcgnueabihf
[G..X]   arm-cortexa9_neon-linux-gnueabihf
[G..X]   x86_64-w64-mingw32,arm-cortexa9_neon-linux-gnueabihf
[G...]   arm-multilib-linux-uclibcgnueabi
[G...]   arm-nano-eabi
[G...]   arm-unknown-eabi
[G...]   arm-unknown-linux-gnueabi
[G..X]   arm-unknown-linux-musleabi
[G...]   arm-unknown-linux-uclibcgnueabi
[G..X]   arm-unknown-linux-uclibcgnueabihf
[G...]   armeb-unknown-eabi
[G...]   armeb-unknown-linux-gnueabi
[G...]   armeb-unknown-linux-uclibcgnueabi
[G...]   armv6-nommu-linux-uclibcgnueabi
[G...]   armv6-rpi-linux-gnueabi
[G...]   armv7-rpi2-linux-gnueabihf
[G...]   armv8-rpi3-linux-gnueabihf
[G...]   avr
[G...]   i586-geode-linux-uclibc
[G...]   i686-centos6-linux-gnu
[G...]   i686-centos7-linux-gnu
[G...]   i686-nptl-linux-gnu
[G...]   i686-ubuntu12.04-linux-gnu
[G...]   i686-ubuntu14.04-linux-gnu
[G...]   i686-ubuntu16.04-linux-gnu
[G..X]   i686-w64-mingw32
[G...]   m68k-unknown-elf
[G...]   m68k-unknown-uclinux-uclibc
[G...]   powerpc-unknown-linux-uclibc,m68k-unknown-uclinux-uclibc
[G...]   mips-ar2315-linux-gnu
[G...]   mips-malta-linux-gnu
[G...]   mips-unknown-elf
[G...]   mips-unknown-linux-uclibc
[G...]   mips64el-multilib-linux-uclibc
[G...]   mipsel-multilib-linux-gnu
[G...]   mipsel-sde-elf
[G...]   mipsel-unknown-linux-gnu
[G..X]   moxie-unknown-elf
[G..X]   x86_64-multilib-linux-uclibc,moxie-unknown-moxiebox
[G..X]   moxiebox
[G..X]   msp430-unknown-elf
[G...]   nios2-altera-linux-gnu
[G..X]   i686-w64-mingw32,nios2-spico-elf
[G...]   nios2-unknown-elf
[G...]   powerpc-405-linux-gnu
[G...]   powerpc-8540-linux-gnu
[G...]   powerpc-860-linux-gnu
[G...]   powerpc-e300c3-linux-gnu
[G...]   powerpc-e500v2-linux-gnuspe
[G...]   x86_64-multilib-linux-uclibc,powerpc-unknown-elf
[G...]   powerpc-unknown-linux-gnu
[G...]   powerpc-unknown-linux-uclibc
[G...]   powerpc-unknown_nofpu-linux-gnu
[G...]   powerpc64-multilib-linux-gnu
[G...]   powerpc64-unknown-linux-gnu
[G...]   powerpc64le-unknown-linux-gnu
[G..X]   riscv32-hifive1-elf
[G..X]   riscv32-unknown-elf
[G..X]   riscv64-unknown-elf
[G..X]   riscv64-unknown-linux-gnu
[G..X]   s390-ibm-linux-gnu
[G...]   s390x-ibm-linux-gnu
[G...]   sh-multilib-linux-gnu
[G...]   sh-multilib-linux-uclibc
[G...]   sh-unknown-elf
[G...]   sparc-leon-linux-uclibc
[G...]   sparc-unknown-linux-gnu
[G...]   sparc64-multilib-linux-gnu
[G...]   x86_64-centos6-linux-gnu
[G...]   x86_64-centos7-linux-gnu
[G...]   x86_64-multilib-linux-gnu
[G..X]   x86_64-multilib-linux-musl
[G...]   x86_64-multilib-linux-uclibc
[G..X]   x86_64-w64-mingw32,x86_64-pc-linux-gnu
[G...]   x86_64-ubuntu12.04-linux-gnu
[G...]   x86_64-ubuntu14.04-linux-gnu
[G...]   x86_64-ubuntu16.04-linux-gnu
[G...]   x86_64-unknown-linux-gnu
[G...]   x86_64-unknown-linux-uclibc
[G..X]   x86_64-w64-mingw32
[G..X]   xtensa-fsf-elf
[G...]   xtensa-fsf-linux-uclibc
 L (Local)       : sample was found in current directory
 G (Global)      : sample was installed with crosstool-NG
 X (EXPERIMENTAL): sample may use EXPERIMENTAL features
 B (BROKEN)      : sample is currently broken
 O (OBSOLETE)    : sample needs to be upgraded
~/Downloads asmirnov
> ct-ng version
This is crosstool-NG version 1.24.0

Copyright (C) 2008  Yann E. MORIN <yann.morin.1998@free.fr>
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
```

jcmvbkbc, 2020-06-13:
You can try a canadian cross with host = arm and target = xtensa based on the existing lx106 configuration. If you build it statically it has a good chance of running on android.

closed 2020-06-13 by 4ntoine