jcmvbkbc/crosstool-NG #68: Can't compile for arm host

4ntoine, 2020-06-15:
assume using the following:
* android-ndk-r21 - standalone toolchain installed with `--arch arm --api 21 --install-dir ~/x-tools/arm-ndk-21` and `~/x-tools/arm-ndk-21/bin` is added to `PATH`.
* gcc version updated to 9.2.0 (tried 8.3.0 too)
* isl version 0.18
* cloog version 0.18.4

created "arm-ndk,xtensa-lx106-elf" sample:
```
# created by 4ntoine
# this comes from samples/xtensa-lx106:
# ---
CT_EXPERIMENTAL=y
CT_PREFIX_DIR="${CT_TOP_DIR}/builds/arm-ndk-${CT_TARGET}"
CT_PATCH_BUNDLED_LOCAL=y
CT_LOCAL_PATCH_DIR="${CT_TOP_DIR}/local-patches"
CT_ARCH_xtensa=y
CT_XTENSA_CUSTOM=y
CT_ARCH_XTENSA_CUSTOM_NAME="lx106"
CT_ARCH_XTENSA_CUSTOM_OVERLAY_LOCATION="${CT_TOP_DIR}/overlays"
CT_TARGET_VENDOR="lx106"
CT_LIBC_NEWLIB_V_2_0_0=y
# CT_LIBC_NEWLIB_ENABLE_TARGET_OPTSPACE is not set
CT_CC_LANG_CXX=y
# CT_CC_STATIC_LIBSTDCXX is not set
CT_DEBUG_gdb=y
# CT_GDB_CROSS_PYTHON is not set
CT_GDB_V_7_5_1=y
# ---
# added by 4ntoine to make canadian cross linked statically:
CT_ISL_V_0_18_0=y
CT_CC_V_8_3_0=y
CT_ISL_VERSION="0.18"
CT_CLOOG_VERSION="0.18.4"
CT_CC_VERSION="9.2.0"
CT_CANADIAN=y
CT_HOST="arm-linux-androideabi"
CT_CONFIGURE_has_static_link=y
CT_WANTS_STATIC_LINK=y
CT_WANTS_STATIC_LINK_CXX=y
CT_STATIC_TOOLCHAIN=y
CT_STRIP_HOST_TOOLCHAIN_EXECUTABLES=y
CT_TOOLCHAIN_TYPE="cross"
``` and created configuration from it (attached).

It results into the following:
```
checking for the value of EOF... configure: error: computing EOF failed
make[2]: ** [configure-target-libstdc++-v3] Error 1
```

4ntoine, 2020-06-15:
[config.zip](https://github.com/jcmvbkbc/crosstool-NG/files/4779465/config.zip)

4ntoine, 2020-06-15:
[build.zip](https://github.com/jcmvbkbc/crosstool-NG/files/4779471/build.zip)

4ntoine, 2020-06-15:
`CT_TOOLCHAIN_TYPE="cross"` seems to be wrong, trying `CT_TOOLCHAIN_TYPE="canadian"..`

4ntoine, 2020-06-15:
Nope, did not help:

[config.zip](https://github.com/jcmvbkbc/crosstool-NG/files/4779569/config.zip)
[build.zip](https://github.com/jcmvbkbc/crosstool-NG/files/4779570/build.zip)

4ntoine, 2020-06-15:
Same errof if trying with `CT_HOST="arm-unknown-linux-gnueabi"` (and toolchain compiled with upstream crosstool-ng, installed and added to PATH)

nolange, 2020-06-15:
Dont think this is related to #1188, do a google search on "checking for the value of EOF", I think you have some directories mixed up.
look into the config.log in `/opt/Espressif/crosstool-NG-android/.build/xtensa-lx106-elf/build/build-cc-final-build-x86_64-build_unknown-linux-gnu`for more clues.

4ntoine, 2020-06-15:
@nolange i did googled before obviously, but unfortunately at the moment i don't have enough understanding on what's wrong and how to fix it. Will continue, any advice is appreciated

4ntoine, 2020-06-15:
hmm, `config.log` in `/opt/Espressif/crosstool-NG-android/.build/xtensa-lx106-elf/build/build-cc-final-build-x86_64-build_unknown-linux-gnu` has `configure: exit 0`, no mentions of "EOF issue there"

4ntoine, 2020-06-16:
this is from "/opt/Espressif/crosstool-NG-android/.build/xtensa-lx106-elf/build/build-cc-final-build-x86_64-build_unknown-linux-gnu/xtensa-lx106-elf/libstdc++-v3/config.log" due to missing header:
```
...
configure:20638:  /opt/Espressif/crosstool-NG-android/.build/xtensa-lx106-elf/build/build-cc-final-build-x86_64-build_unknown-linux-gnu/./gcc/xgcc -B/opt/Espressif/crosstool-NG-android/.build/xtensa-lx106-elf/build/build-cc-final-build-x86_64-build_unknown-linux-gnu/./gcc/ -B/opt/Espressif/crosstool-NG-android/.build/xtensa-lx106-elf/buildtools/xtensa-lx106-elf/bin/ -B/opt/Espressif/crosstool-NG-android/.build/xtensa-lx106-elf/buildtools/xtensa-lx106-elf/lib/ -isystem /opt/Espressif/crosstool-NG-android/.build/xtensa-lx106-elf/buildtools/xtensa-lx106-elf/include -isystem /opt/Espressif/crosstool-NG-android/.build/xtensa-lx106-elf/buildtools/xtensa-lx106-elf/sys-include    -c -g -O2 -g -Os  conftest.c >&5
In file included from /opt/Espressif/crosstool-NG-android/.build/xtensa-lx106-elf/buildtools/xtensa-lx106-elf/include/_ansi.h:16,
                 from /opt/Espressif/crosstool-NG-android/.build/xtensa-lx106-elf/buildtools/xtensa-lx106-elf/include/stdio.h:29,
                 from conftest.c:14:
/opt/Espressif/crosstool-NG-android/.build/xtensa-lx106-elf/buildtools/xtensa-lx106-elf/include/sys/config.h:170:10: fatal error: xtensa/config/core-isa.h: No such file or directory
  170 | #include <xtensa/config/core-isa.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
configure:20638: $? = 1
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "package-unused"
| #define PACKAGE_TARNAME "libstdc++"
| #define PACKAGE_VERSION "version-unused"
| #define PACKAGE_STRING "package-unused version-unused"
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL ""
| #define LT_OBJDIR ".libs/"
| #define _GLIBCXX_HOSTED 1
| #define _GLIBCXX_VERBOSE 1
| #define _GLIBCXX_USE_LONG_LONG 1
| #define _GLIBCXX_FULLY_DYNAMIC_STRING 0
| /* end confdefs.h.  */
| #include <stdio.h>
| int
| main ()
| {
| static int test_array [1 - 2 * !((EOF) < 0)];
| test_array [0] = 0;
| return test_array [0];
| 
|   ;
|   return 0;
| }
configure:20641: error: computing EOF failed
...
```

4ntoine, 2020-06-18:
Question on SO: https://stackoverflow.com/questions/62457315/how-to-make-canadian-cross-for-xtensa-lx106-elf-for-android-arm-host-with-crosst

jcmvbkbc, 2020-06-20:
I've pushed a branch xtensa-1.24.x with a defconfig aarch64-linux-gnu,xtensa-lx106-elf that I used to build a static toolchain for aarch64-linux host, xtensa-lx106-elf target. Please give it a try.

4ntoine, 2020-06-20:
Hi, jcmvbkbc. Thanks for looking into it!

I had to change host to "aarch64-linux-android" to use it with standalone toolchain from Android NDK 21b (refeicev using "./build/tools/make_standalone_toolchain.py --arch arm64 --api 21 --install-dir /home/anton/x-tools/arm64-ndk").

[config.zip](https://github.com/jcmvbkbc/crosstool-NG/files/4808113/config.zip)

Eventually i get the following:
```
[INFO ]  =================================================================
[INFO ]  Installing gettext for host
[EXTRA]    Configuring gettext
[EXTRA]    Building gettext
[ERROR]    /opt/Espressif/crosstool-NG/.build/HOST-aarch64-linux-android/xtensa-lx106-elf/src/gettext/gettext-tools/gnulib-lib/spawn_faction_addclose.c:50:19: error: member reference base type 'posix_spawn_file_actions_t' (aka 'struct __posix_spawn_file_actions *') is not a structure or union
[ERROR]    /opt/Espressif/crosstool-NG/.build/HOST-aarch64-linux-android/xtensa-lx106-elf/src/gettext/gettext-tools/gnulib-lib/spawn_faction_addclose.c:50:42: error: member reference base type 'posix_spawn_file_actions_t' (aka 'struct __posix_spawn_file_actions *') is not a structure or union
[ERROR]    /opt/Espressif/crosstool-NG/.build/HOST-aarch64-linux-android/xtensa-lx106-elf/src/gettext/gettext-tools/gnulib-lib/spawn_faction_addclose.c:59:24: error: member reference base type 'posix_spawn_file_actions_t' (aka 'struct __posix_spawn_file_actions *') is not a structure or union
[ERROR]    /opt/Espressif/crosstool-NG/.build/HOST-aarch64-linux-android/xtensa-lx106-elf/src/gettext/gettext-tools/gnulib-lib/spawn_faction_addclose.c:59:47: error: member reference base type 'posix_spawn_file_actions_t' (aka 'struct __posix_spawn_file_actions *') is not a structure or union
[ERROR]    /opt/Espressif/crosstool-NG/.build/HOST-aarch64-linux-android/xtensa-lx106-elf/src/gettext/gettext-tools/gnulib-lib/spawn_faction_addclose.c:64:19: error: member reference base type 'posix_spawn_file_actions_t' (aka 'struct __posix_spawn_file_actions *') is not a structure or union
[ERROR]    /opt/Espressif/crosstool-NG/.build/HOST-aarch64-linux-android/xtensa-lx106-elf/src/gettext/gettext-tools/gnulib-lib/spawn_faction_adddup2.c:50:19: error: member reference base type 'posix_spawn_file_actions_t' (aka 'struct __posix_spawn_file_actions *') is not a structure or union
[ERROR]    /opt/Espressif/crosstool-NG/.build/HOST-aarch64-linux-android/xtensa-lx106-elf/src/gettext/gettext-tools/gnulib-lib/spawn_faction_adddup2.c:50:42: error: member reference base type 'posix_spawn_file_actions_t' (aka 'struct __posix_spawn_file_actions *') is not a structure or union
[ERROR]    /opt/Espressif/crosstool-NG/.build/HOST-aarch64-linux-android/xtensa-lx106-elf/src/gettext/gettext-tools/gnulib-lib/spawn_faction_adddup2.c:59:24: error: member reference base type 'posix_spawn_file_actions_t' (aka 'struct __posix_spawn_file_actions *') is not a structure or union
[ERROR]    /opt/Espressif/crosstool-NG/.build/HOST-aarch64-linux-android/xtensa-lx106-elf/src/gettext/gettext-tools/gnulib-lib/spawn_faction_adddup2.c:59:47: error: member reference base type 'posix_spawn_file_actions_t' (aka 'struct __posix_spawn_file_actions *') is not a structure or union
[ERROR]    /opt/Espressif/crosstool-NG/.build/HOST-aarch64-linux-android/xtensa-lx106-elf/src/gettext/gettext-tools/gnulib-lib/spawn_faction_adddup2.c:65:19: error: member reference base type 'posix_spawn_file_actions_t' (aka 'struct __posix_spawn_file_actions *') is not a structure or union
[ERROR]    make[6]: *** [Makefile:2962: spawn_faction_addclose.lo] Error 1
[ERROR]    make[6]: *** Waiting for unfinished jobs....
[ERROR]    make[6]: *** [Makefile:2962: spawn_faction_adddup2.lo] Error 1
[ERROR]    /opt/Espressif/crosstool-NG/.build/HOST-aarch64-linux-android/xtensa-lx106-elf/src/gettext/gettext-tools/gnulib-lib/spawn_faction_init.c:34:30: error: member reference base type 'posix_spawn_file_actions_t' (aka 'struct __posix_spawn_file_actions *') is not a structure or union
[ERROR]    /opt/Espressif/crosstool-NG/.build/HOST-aarch64-linux-android/xtensa-lx106-elf/src/gettext/gettext-tools/gnulib-lib/spawn_faction_init.c:35:39: error: member reference base type 'posix_spawn_file_actions_t' (aka 'struct __posix_spawn_file_actions *') is not a structure or union
[ERROR]    /opt/Espressif/crosstool-NG/.build/HOST-aarch64-linux-android/xtensa-lx106-elf/src/gettext/gettext-tools/gnulib-lib/spawn_faction_init.c:42:15: error: member reference base type 'posix_spawn_file_actions_t' (aka 'struct __posix_spawn_file_actions *') is not a structure or union
[ERROR]    /opt/Espressif/crosstool-NG/.build/HOST-aarch64-linux-android/xtensa-lx106-elf/src/gettext/gettext-tools/gnulib-lib/spawn_faction_init.c:43:15: error: member reference base type 'posix_spawn_file_actions_t' (aka 'struct __posix_spawn_file_actions *') is not a structure or union
[ERROR]    make[6]: *** [Makefile:2962: spawn_faction_init.lo] Error 1
[ERROR]    /opt/Espressif/crosstool-NG/.build/HOST-aarch64-linux-android/xtensa-lx106-elf/src/gettext/gettext-tools/gnulib-lib/spawn_faction_addopen.c:51:19: error: member reference base type 'posix_spawn_file_actions_t' (aka 'struct __posix_spawn_file_actions *') is not a structure or union
[ERROR]    /opt/Espressif/crosstool-NG/.build/HOST-aarch64-linux-android/xtensa-lx106-elf/src/gettext/gettext-tools/gnulib-lib/spawn_faction_addopen.c:51:42: error: member reference base type 'posix_spawn_file_actions_t' (aka 'struct __posix_spawn_file_actions *') is no [... cut ...]
[ERROR]  >>  Make sure your report includes all the information pertinent to this issue.
[ERROR]  >>  Read the bug reporting guidelines here:
[ERROR]  >>      http://crosstool-ng.github.io/support/
[ERROR]   
[ERROR]  (elapsed: 9:47.10)
[09:48] / make: *** [ct-ng:261: build] Error 2
```
[build.zip](https://github.com/jcmvbkbc/crosstool-NG/files/4808114/build.zip)

What toolchain should be used for "aarch64-linux-gnu"? I've tried "./ct-ng aarch64-linux-gnu" but it was not found. Am i missing anything else?

4ntoine, 2020-06-20:
Worked for me if compiling with following ubuntu packages (NOT from NDK):
```
sudo apt install g++-aarch64-linux-gnu
sudo apt install gcc-aarch64-linux-gnu
```

closed 2020-06-24 by 4ntoine