jcmvbkbc/crosstool-NG #69: Error: invalid register 'atomctl' for 'wsr' instruction in reset-vector.S

sunilbage, 2020-08-08:
Getting an error while cross compling xtensa source xtos  with crosstool-NG 
reset-vector.S: Assembler messages:
reset-vector.S:291: Error: unknown opcode or format name 'wsr.atomctl'
Makefile.src:281: recipe for target 'reset-vector-unpack.o' failed
make: *** [reset-vector-unpack.o] Error 1
Host : 32 bit ubuntu 16.04 LTS
crosstool-NG : crosstool-NG 1.24.0

jcmvbkbc, 2020-08-08:
> Getting an error while cross compling xtensa source xtos with crosstool-NG

I don't think you can compile xtos with the crosstool-NG. Perhaps you're compiling xtos with the compiler that you've built with the crosstool-NG.

> Error: unknown opcode or format name 'wsr.atomctl'

That means that xtensa configuration that you used to generate your compiler does not have atomctl register.
Either this is a wrong xtensa configuration, or xtos does not properly check for the presence of atomctl.

In any case I don't see how this may be a crosstool-NG issue.

sunilbage, 2020-08-09:
Yes you are right. I am compiling xtos with the compiler that you've built with the crosstool-NG. Xtensa overlay file which I am using for custom xtensa configuration is having atomctrl register.
I would like to know that if I have missed any configuration parameters while cross compiling the compiler to o support atomctrl register?.
Or how to enable atomctrl register support for the compiler while cross compiling it?
Please help me with this.

jcmvbkbc, 2020-08-09:
> Please help me with this.

Ok, please share your crosstool-NG .config that you used to build the compiler and your xtensa configuration overlay.

> Or how to enable atomctrl register support for the compiler while cross compiling it?

You need to build binutils with the correct configuration overlay.

sunilbage, 2020-08-09:
I am attaching the crosstool-NG .config that you used to build the compiler for your reference. Pls let me me know what changes i have to do to build binutils with the correct configuration overlay.
[crosstool-NG.config.zip](https://github.com/jcmvbkbc/crosstool-NG/files/5046791/crosstool-NG.config.zip)

jcmvbkbc, 2020-08-09:
>> Ok, please share your crosstool-NG .config that you used to build the compiler and your xtensa configuration overlay.

> I am attaching the crosstool-NG .config that you used to build the compiler for your reference.

Please also share the configuration overlay, the file ${CT_TOP_DIR}/overlay/xtensa-lg_108_lx7_x*

sunilbage, 2020-08-09:
Sir the overlay size is huge. around 1.9 GB. It is difficult for me to upload the huge file. Even after zipping the overlay file.
Can u suggest if any error exists with the crosstool-NG .config file which i uploaded earlier.

jcmvbkbc, 2020-08-09:
> the overlay size is huge. around 1.9 GB

This is hardly possible. [Here's](https://github.com/jcmvbkbc/xtensa-toolchain-build/tree/master/overlays) an example collection of overlay files, the biggest one is smaller than 1MB. The overlay file used by the ct-ng can be generated from the original overlay, normally installed into XtDevTools/install/builds/<xtensa-tools-release>/<core-name>/src/xtensa-config-overlay.tar.gz by [this](https://github.com/jcmvbkbc/xtensa-toolchain-build/blob/master/split-overlay.sh) script.

sunilbage, 2020-08-09:
Ahhha. I got my mistake. I was giving the wrong overlay file. I missed the step you mentioned. I will try again by giving the correct overlay file and cross compile the compiler and binutils again and check.
If i still face the issue. I will attach the overlay file. 
Thanks for the suggestion.

sunilbage, 2020-08-09:
Still after using the correct overlay file i am getting the same above error.
xtensa-lg_108_lx7_x-elf-gcc -c -O2 -g -mlongcalls -Wall -Werror  -I/home/sunil/XtDevTools/install/tools/RG-2019.12-win32/XtensaTools/xtensa-lg_108_lx7_x-elf/include -I/home/sunil/XtDevTools/install/builds/RG-2019.12-win32/lg_108_lx7_x/xtensa-lg_108_lx7_x-elf/arch/include -I/home/sunil/XtDevTools/install/builds/RG-2019.12-win32/lg_108_lx7_x/xtensa-lg_108_lx7_x-elf/include  -mtext-section-literals -DXTOS_UNPACK -o reset-vector-unpack.o reset-vector.S
reset-vector.S: Assembler messages:
reset-vector.S:291: Error: unknown opcode or format name 'wsr.atomctl'
Makefile.src:281: recipe for target 'reset-vector-unpack.o' failed
make: *** [reset-vector-unpack.o] Error 1

I have attached the overlay file as you have asked for. Please find the xtensa_lg_108_lx7_x.tar.gz file attached.
[xtensa_lg_108_lx7_x.tar.gz](https://github.com/jcmvbkbc/crosstool-NG/files/5047423/xtensa_lg_108_lx7_x.tar.gz)

jcmvbkbc, 2020-08-09:
I've put your overlay into ```${CT_TOP_DIR}/overlay/``` and built the toolchain with ct-ng 1.24.0 using your original config file. It works for me:
```
$ cat <<EOF > test.S
>         .text
>         wsr     a2, atomctl
> EOF
$ /home/jcmvbkbc/x-tools/xtensa-lg_108_lx7_x-elf/bin/xtensa-lg_108_lx7_x-elf-gcc -c test.S 
$ /home/jcmvbkbc/x-tools/xtensa-lg_108_lx7_x-elf/bin/xtensa-lg_108_lx7_x-elf-objdump -d test.o 

test.o:     file format elf32-xtensa-le

Disassembly of section .text:

00000000 <.text>:
   0:   136320          wsr.atomctl     a2
```

Please make sure that you use the correct toolchain when building xtos?