jcmvbkbc/crosstool-NG #70: crosstool-NG cross compiled gcc is giving error while compiling

sunilbage, 2020-08-18:
Dear Sir,
I have using cross-compiled gcc by Crosstool-NG for Xtensa.
While compiling Xtensa sources i am getting error 
xtensa-<customized-elf>-gcc: error: unrecognized command line option '-mno-coproc'; did you mean '-fno-coomon' ?
xtensa-<customized-elf>-gcc: error: unrecognized command line option '-mno-l32r-flix

can you suggest me how to resolve this issue.
Have i missed any configuration parameters while configuring cross-tool-NG?

jcmvbkbc, 2020-08-18:
> can you suggest me how to resolve this issue.

Yes. Edit makefiles of the project you're trying to build and remove options `-mno-coproc` and `-mno-l32r-flix` from the gcc command line. They are not supported by xtensa gcc.

- `-mno-l32r-flix` removal does not change anything because xtensa gcc does not use FLIX bundles.
- `-mno-coproc` removal means that xtensa gcc would use FPU for floating point calculations if it's available in the configured xtensa core. There's currently no way to disable use of FPU from the command line. xtensa gcc does not use any other coprocessor.

sunilbage, 2020-08-18:
Dear Sir,
After removing the options -mno-coproc and -mno-l32r-flix from the gcc command line as per your suggestion i am able to proceed with the compilation.
But now proceeding ahead, i am encountering another error with respect to assembly flags
ASFLAGS       = -O2 -Os -g $(CMNFLAGS) -Wa,--schedule

unrecognized option '--schedule' error is thrown by xtensa--gcc.
Please suggest me how to resolve this issue.

sunilbage, 2020-08-18:
Sir,
FLIX instructions are supported by Xtensa Processor as per the below link.

https://ip.cadence.com/news/76/330/Tensilica-Previews-Next-Generation-Xtensa-ISA-With-Flexible-Length-Instructions-At-Microprocessor-Forum

My query is if we remove the FLIX option, then  compiled Xtensa sources will not have the support of executing FLIX instructions.
This is not we want , right??

I am encountering one more error with respect to ASFLAGS during compilation
ASFLAGS       = -O2 -Os -g $(CMNFLAGS) -Wa,--schedule
 unrecognized option '--schedule' error during compilation. Please suggest how to resolve this issue also.

jcmvbkbc, 2020-08-18:
> unrecognized option '--schedule' error is thrown by xtensa--gcc.
> Please suggest me how to resolve this issue.

Just the same: remove it from the compiler command line. Mainline version of xtensa assembler does not support it.

> My query is if we remove the FLIX option, then compiled Xtensa sources will not have the support of executing FLIX instructions.

No. You cannot take support for executing FLIX instruction out of processor. It will run FLIX instructions as always.
But the mainline xtensa gcc doesn't generate FLIX instructions and the mainline xtensa assembler has no ability to schedule multiple opcodes into a FLIX instruction.

sunilbage, 2020-08-19:
Sir ,
after removing the --schedule  from the compiler command line, compilation is not proceeding further. It is stuck compiling the assembly file.S.
What could be the reason  and how to resolve it?

jcmvbkbc, 2020-08-19:
The whole `-Wa,--schedule` construct must be removed, not just the `--schedule` part.

sunilbage, 2020-08-20:
My issues are all resolved.
Thank You.

closed 2020-08-20 by sunilbage