jcmvbkbc/crosstool-NG #71: Toolchain for ARM produces corrupted files

4ntoine, 2021-01-21:
When compiled for ARM 32bit `CT_HOST=arm-linux-gnueabihf` archiever fails to link `.o` files into `.a` with return code 159. I will provide the details later. If not archiving .o into .a and pass to linker linker fails to link with the same return code.

Is it a known issue?

100% the same command-lines that use toolchain compiled for arm64 (`CT_HOST=aarch64-linux-gnu`) works just fine.

jcmvbkbc, 2021-01-21:
I know that there's been a user who ran the toolchain on armhf host, they've reported a real issue in the compiler, but after it was fixed they were happy, so presumably it worked for them. I guess it doesn't happen with arbitrary `.o` files? I'll try to reproduce it anyway.

> I will provide the details later.

That would be helpful, thank you. Please provide the version of ct-ng and configuration used to build the toolchan as well as the toolchain invocation commands that lead to the crash.

4ntoine, 2021-01-21:
Here is how the extensa toolchain is built.

Common
-----

git clone https://github.com/jcmvbkbc/crosstool-NG.git
cd crosstool-NG.git
git checkout xtensa-1.24.x
./bootstrap
./configure --prefix=./installed
make
make install
cd ./installed/bin

ARM64 (works fine)
-----

sudo apt install g++-aarch64-linux-gnu
sudo apt install gcc-aarch64-linux-gnu

mkdir overlays
download https://github.com/jcmvbkbc/crosstool-NG/blob/xtensa-1.22.x/overlays/xtensa_lx106.tar to overlays (and extract).
./ct-ng list-samples
./ct-ng aarch64-linux-gnu,xtensa-lx106-elf
./ct-ng build.4

Grab the files from ~/x-tools/HOST-aarch64-linux-gnu

ARM32 (bad instructions)
-----

sudo apt install gcc-arm-linux-gnueabihf
sudo apt install g++-arm-linux-gnueabihf

edit .config and set CT_HOST=arm-linux-gnueabihf
./ct-ng build.4

Grab the files from ~/x-tools/HOST-arm-linux-gnueabihf.

The files
----

Compiled toolchain for ARM, SDK distrib (including compiled toolchain), screenshot and build .b files for xtensa in:
https://dropmefiles.com/SFH4q
Note README for the explanation.

Compilation log (how to reproduce):
https://dropmefiles.com/2nSaW

> bin/xtensa-lx106-elf-ar cru /data/data/../coreBuild/core.a /data/data/../coreBuild/cont.S.o
xtensa-lx106-elf-ar returned 159

4ntoine, 2021-01-21:
@jcmvbkbc  Let me know if you need anything else to repro.

4ntoine, 2021-01-21:
Yup, just compile any `.cpp` to `.o` and archieve it to `.a` and it fails with return code 159:
https://dropmefiles.com/uShD4

4ntoine, 2021-01-22:
@jcmvbkbc Is there anything i can assist you with?

jcmvbkbc, 2021-01-22:
I don't have easy access to old ARM hardware, I tried your instructions in qemu linux-user, it worked for me. Looking through the emulated instructions trace I've noticed that the following VFP instructions were used:
```
vand
vceq.i8
vdup.8
vld1.8
vldr
vmov
vmov.32
vmov.f32
vpadd.i8
vpop
vpush
vst1.8
vstmia
vstr
```

@4ntoine does you hardware support these instructions? Do you have any way to check whether ar crash on your hardware happens on one of these instructions? Maybe run your reproducer in the debugger or under `strace -i`?

jcmvbkbc, 2021-01-22:
> xtensa-lx106-elf-ar returned 159

OTOH this seems to correspond to termination by SIGSYS. In any case running reproducer on hardware under `strace -i` should shed some light.

4ntoine, 2021-01-22:
@jcmvbkbc Thanks for looking into it.

> does you hardware support these instructions? Do you have any way to check whether ar crash on your hardware happens on one of these instructions?

At the moment i have 1 device for the testing - it's Huawei Honor 9s device, can you find the specs?

I'm afraid `strace` requires root access and my device is not rooted.

I will try to get the stacktrace. BTW i can see the binaries are stripped (in ~/x-tools/..). Any chance to get them unstripped (probably from build dir)? Will it help if you check ARM64 sys calls and compared them (i can provide distrib similar to what i've already uploaded for arm32)?

jcmvbkbc, 2021-01-22:
> At the moment i have one 32bit cpu device for the testing - it's Huawei Honor 9s device, can you find the specs?

We can approach it from the other side: could you use `arm-linux-gnueabi` host triplet and see if the toolchain built with it works?

4ntoine, 2021-01-22:
> > At the moment i have one 32bit cpu device for the testing - it's Huawei Honor 9s device, can you find the specs?
> 
> We can approach it from the other side: could you use `arm-linux-gnueabi` host triplet and see if the toolchain built with it works?

Yup, will do that and return back to you soon

jcmvbkbc, 2021-01-22:
> I'm afraid strace requires root access and my device is not rooted.

Can you try running `ulimit -c unlimited` before running ar, collect the core dump and load it together with ar into cross gdb (`arm-linux-gnueabihf-gdb -c core xtensa-lx106-elf-ar`)?

jcmvbkbc, 2021-01-22:
> At the moment i have only Honor 9s device, can you find the specs?

What I find says that's based on ARM cortex-a53, which is 64-bit ARMv8. Is there a reason to tun 32-bit code on it?
Could you also check `/proc/cpuinfo` contents on your devices?

4ntoine, 2021-01-22:
> > At the moment i have only Honor 9s device, can you find the specs?
> 
> What I find says that's based on ARM cortex-a53, which is 64-bit ARMv8. Is there a reason to tun 32-bit code on it?

Nope, it's detected as 32 bit by Android:
https://stackoverflow.com/questions/65769808/device-is-incorrectly-detected-as-32bit-by-android

It's armv8l, not armv8a. I'm not an expert here but it seems it makes a difference.

> Could you also check `/proc/cpuinfo` contents on your devices?

will do that

> ulimit

I believe it's just absent on "regular android device". Doesn't it require root too?

4ntoine, 2021-01-22:
```
adb shell ls -l /proc/cpuinfo
-r--r--r-- 1 root root 0 2021-01-22 23:29 /proc/cpuinfo
```

can't access it without rooting

4ntoine, 2021-01-22:
[image]

4ntoine, 2021-01-22:
No stacktrace/addressed in the logcat, will try not hard float toolchain

jcmvbkbc, 2021-01-22:
> can't access it without rooting

try `adb shell cat /proc/cpuinfo` instead.

4ntoine, 2021-01-22:
> adb shell cat /proc/cpuinfo

```
processor	: 0
model name	: ARMv8 Processor rev 4 (v8l)
BogoMIPS	: 26.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt lpae evtstrm aes pmull sha1 sha2 crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv8 Processor rev 4 (v8l)
BogoMIPS	: 26.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt lpae evtstrm aes pmull sha1 sha2 crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv8 Processor rev 4 (v8l)
BogoMIPS	: 26.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt lpae evtstrm aes pmull sha1 sha2 crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv8 Processor rev 4 (v8l)
BogoMIPS	: 26.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt lpae evtstrm aes pmull sha1 sha2 crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 4
model name	: ARMv8 Processor rev 4 (v8l)
BogoMIPS	: 26.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt lpae evtstrm aes pmull sha1 sha2 crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 5
model name	: ARMv8 Processor rev 4 (v8l)
BogoMIPS	: 26.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt lpae evtstrm aes pmull sha1 sha2 crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 6
model name	: ARMv8 Processor rev 4 (v8l)
BogoMIPS	: 26.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt lpae evtstrm aes pmull sha1 sha2 crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 7
model name	: ARMv8 Processor rev 4 (v8l)
BogoMIPS	: 26.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt lpae evtstrm aes pmull sha1 sha2 crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: MT6762V/WR

```

4ntoine, 2021-01-23:
`arm-linux-gnueabi` does not work too: same error code 159

4ntoine, 2021-01-23:
@jcmvbkbc Nope, i've just tried without symlinking and targeting Android 28
[image]

same issue.

4ntoine, 2021-01-23:
@jcmvbkbc anything i can try locally just to have more info to decide on? step-by-step  for non-rooted device guide appreciated

4ntoine, 2021-01-29:
Any progress on it? Is there anything i can help with?

jcmvbkbc, 2021-01-29:
I'm setting things up for debugging on android. Expecting to get to it this weekend.

4ntoine, 2021-01-29:
@jcmvbkbc Great! Thank you

jcmvbkbc, 2021-01-31:
I just tried `xtensa-lx106-elf-ar cru z.a BearSSLHelpers.cpp.o` in 32-bit arm android emulator with clean android-10.0.0_r47 build from AOSP. It works when I do it as a root. But I couldn't start xtensa-lx106-elf-ar as a non-root.

How do you set up permissions and security context for the toolchain?

4ntoine, 2021-01-31:
@jcmvbkbc I don't do it and it works with regular apps permissions on arm 64 bit (armv8a).

4ntoine, 2021-01-31:
@jcmvbkbc BTW make sure you target android 9 for testing. Android 10 has more strict security model (thus that symlink trick, fore more infor about the trick [here](https://stackoverflow.com/questions/64786837/what-path-to-put-executable-to-run-on-android-29))

jcmvbkbc, 2021-01-31:
Ok, after disabling selinux with `setenforce 0` non-root user was able to run `xtensa-lx106-elf-ar cru z.a BearSSLHelpers.cpp.o` successfully.
I've also noticed that I could run commands under strace as non-root user with selinux enabled. Would you like to try it?

4ntoine, 2021-01-31:
@jcmvbkbc sure, please provide set-by-step guide on how to do it. My main concern is that even if it works, there is some difference between 32 and 84 toolchain that makes a difference

jcmvbkbc, 2021-01-31:
> My main concern is that even if it works, there is some difference between 32 and 84 toolchain that makes a difference

Maybe. My suggestion is to try to run the reproducer under the strace to see where exactly it fails.
I guess that doing `adb push strace /data/local/tmp` with the strace binary available [here](http://jcmvbkbc.spb.ru/~dumb/tmp/202101310237/strace) followed by `adb shell` and `chmod +x /data/local/tmp/strace` should give you working strace.

jcmvbkbc, 2021-01-31:
I've put xtensa-lx106-elf-ar to /data/local/tmp and `xtensa-lx106-elf-ar cru z.a BearSSLHelpers.cpp.o` runs successfully under non-root with selinux enabled. Looks like this system doesn't reproduce your issue.

4ntoine, 2021-01-31:
I'm not sure how it works but on Android 10 the app can't execute the executables even in it's own app sandbox (`/data/data/%package%`). So if you can execute anything from `data/local/tmp` i'm afraid it does not reproduce the real non-rooted device.

Please provide step-by-step guide what should i try?

jcmvbkbc, 2021-01-31:
> if you can execute anything from data/local/tmp

My understanding is that's because it is specifically designated to the user `shell`, who's also the user that `adb shell` runs under.
I would suggest that you try the same thing: put your reproducer binaries to the /data/local/tmp and run the reproducer from the `adb shell` there. Success would mean that the issue is in the permissions/selinux setup.

4ntoine, 2021-01-31:
@jcmvbkbc That surprisingly worked:
```
> adb shell /data/local/tmp/libesp8266_xtensa-lx106-elf-ar.so cru /data/local/tmp/z.a /data/local/tmp/BearSSLHelpers.cpp.o
/tmp developer
> ls -l /data/local/tmp
ls: /data/local/tmp: No such file or directory
/tmp developer
> adb shell ls -l /data/local/tmp
total 2074
-rw-rw-rw- 1 shell shell  436800 2021-01-31 23:44 BearSSLHelpers.cpp.o
drwxrwxr-x 2 shell shell    3488 2021-01-31 23:44 device-explorer
-rwxrwxrwx 1 shell shell 1234300 2021-01-23 22:48 libesp8266_xtensa-lx106-elf-ar.so
drwxrwxrwx 4 shell shell    3488 2021-01-31 23:22 perfd
-rw-rw-rw- 1 shell shell  439982 2021-01-31 23:48 z.a
```

What does it mean? Why does it work for 64 bit out of box? (from app dir)?

jcmvbkbc, 2021-01-31:
> What does it mean?

That the toolchain binaries that you have are capable of working correctly on your system. My interpretation is that the original issue is in the permissions/selinux setup.

> Why does it work for 64 bit out of box? Can it be done to work the same way on 32 bit?

These questions are about the android and its application permissions of which I don't know much. I would expect that there might be a tool for debugging this kind of issues.

4ntoine, 2021-02-01:
@jcmvbkbc Well, regular android app on Android 10 can't put files to `/data/local/tmp` (i've just checked and got `AccessDeniedException`) so this workaround does not work. That's still strange that some special tricks are required to make it working on 32..

4ntoine, 2021-02-02:
@jcmvbkbc Ok, so what's the plan? I can prepare a simple app that extracts the sdk and compiles the final binary for the testing, if it helps.

jcmvbkbc, 2021-02-02:
> so what's the plan?

I don't have any further plan. Specifically I'm not planning to debug android application permissions/selinux configuration.

You may want to find out specific syscall that gets SIGSYS and figure out why that happens. Like I said I don't know android well enough to give you any guidance on this, the only thing that comes to my mind is that perhaps generic selinux policy debug techniques could be useful. When you know which syscall fails and for what reason you may be able to find a fix or workaround for it.